iNavFlight/inav #1763: Omnibus F4 (non pro)  SBUS not working

Key0nee, 2017-06-05:
Hello,
   I am using an Omnibus F4 (non pro) purchased from RTFQuads... in a small GPS camera ship build.  I am using an Futaba Sbus receiver (R7008SB) connected to UART 6 marked PPM/SBUS and SBUS is shorted to middle pad.  All other required iNav settings are verified (ie port, serial rx type etc...)  No receiver values are making it to the flight controller.  I've also tried latest iNav version for other Omnibus f4 targets with no luck..

I thought it was a bad board but then I tried the latest version of Cleanflight for F4SD target and it worked fine... all values from receiver are being reported in Cleanflight.  Only tested on UART 6.

scrapit85, 2017-06-05:
This is the same issue as #1643 
Also tried to get it to work with Frsky sbus. As you said Betaflight and Cleanflight and the modified Exotic OMNIBUS iNav 1.6.1 work fine with it setup on UART 6

Key0nee, 2017-06-05:
Sorry... I should have read thru here more thoroughly... I already felt out of place posting here and not being a developer...
Thanks for the post... I'll read issue #1643

moggiex, 2017-06-05:
@Key0nee daft one, see this video as you might have 3 pads that 2 of them might need bridging to make SBUS work [https://www.youtube.com/watch?v=ZFVIbU4aJvo](https://www.youtube.com/watch?v=ZFVIbU4aJvo) or at least a sanity check you've done it right as I had the same issue, Matt

jflyper, 2017-06-07:
@all Sorry, it's my bad not submitting PR for this problem. It jumped two notches in my list of todos... 😭

iconicj, 2017-06-08:
I thought this was a known limitation of the f4s only supporting sbus inversion on one uart - uart1, next to the motor output headers.

DzikuVx, 2017-06-08:
@iconicj It's not STM32F4 but rather FC itself problem. SBUS can be connected only to UART that has additional inverter in from of it. It can be any UART or even any number of UARTs

jflyper, 2017-06-09:
@here 
#1779 is submitted, but need some testing.
[inav_1.7.1_OMNIBUSF4V3-0.hex.zip](https://github.com/iNavFlight/inav/files/1064490/inav_1.7.1_OMNIBUSF4V3-0.hex.zip)

I also need someone to test the F4, F4V2 and F4Pro for regression test...

scrapit85, 2017-06-09:
@jflyper Just tested on a F4 V3 board and Frsky the SBUS input is now working.
Also tested on a F4 Pro V2 and the SBUS is still working.

jflyper, 2017-06-09:
@scrapit85 
Thanks for testing! Can you keep the OMNIBUSF3V3 running, possibly with flight test?

scrapit85, 2017-06-09:
Will leave it running for a while then just got to install it in a plane hopefully test tomorrow.

jflyper, 2017-06-09:
@scrapit85 Ah, I meant "can you leave the board loaded with OMNIBUSF3V3" by "keep it running".  Looking forward to hearing the flight test :)

Key0nee, 2017-06-10:
Futaba SBUS working on F4v3...

Key0nee, 2017-06-11:
@jflyper Thanks for the fix my futaba SBUS is working well but I have a new issue (maybe)... Can you comment on weather PR [ #1607](https://github.com/iNavFlight/inav/pull/1607) is rolled up in the firmware you linked above.  there is NO signal to my ESC's on PWM outputs 1&2 but they show up fine on PWM outputs 3&4.  Currently, I don't have a need for servo outputs so I don't have pins soldered to PWM outputs 5&6.  If the PWM outputs are just remaped, is there a simple end user method to remap to stock settings ?

Thank You
K

jflyper, 2017-06-11:
@Key0nee Ah, I think I know what's happening.

jflyper, 2017-06-11:
@Key0nee Your esc/servo mapping issue is completely different from the SBUS problem?

jflyper, 2017-06-12:
@Key0nee Proposed mapping in #1607 is effective only for airplane or some mixer with servos. It shouldn't affect pure multi-rotor vehicles. Oh, are you using servos for a gimbal?

Key0nee, 2017-06-12:
@jflyper No gimbal or servos used... wih inav 1.7.1 installed and using the motor tab to individually test spin motors i get what I expected, motors 1-4 corresponding to PWM outputs 1-4

When using inav 1.7.1 F4V3-0, using the motor tab to individually spin up motors result in no output on PWM 1-2, but motors 1-2 present on PWM outputs 3-4...  i will try get pins installed for PWM outputs 5-6 tomorrow and test again...

jflyper, 2017-06-12:
@Key0nee Ah!!! It's probably my bad... Hang on, I'll investigate and fix it.

jflyper, 2017-06-15:
@Key0nee
Did you try the updated firmware?

Key0nee, 2017-06-15:
@jflyper I just downloaded... been busy at work and haven't had time but will try and report tonight...

Key0nee, 2017-06-16:
@jflyper installed and initial power up all look good... SBUS works great and motor shift issue is gone.. will continue to update... Thanks

jflyper, 2017-06-16:
@Key0nee Thanks for testing & reporting!

Key0nee, 2017-06-18:
@jflyper  just tested the LED strip control... No Bueno... [Link](https://github.com/iNavFlight/inav/issues/1803)

closed 2017-06-20 by Key0nee