iNavFlight/inav #1764: V-Tail210 - SP Racing F3 Evo -Accelerometer - Magnetometer / OSD not available

yo8aiv, 2017-06-05:
While Gps works great and there are great options 
V-Tail210 - SP Racing F3 Evo - Accelerometer -Magnetometer / OSD not available 

not detected correctly and when i select still its not taken into account

GYRO=MPU9250, ACC=MPU9250 - works in cleanflight
So there must be some conflicts
OSD not being detected .

iNav status below:
GYRO=MPU6500, ACC=MPU6500, BARO=BMP280, MAG=AK8963
Sensor status: GYRO=OK, ACC=OK, MAG=OK, BARO=OK, SONAR=NONE, GPS=UNAVAILABLE
If needed i can make tests.
Thanks.

stronnag, 2017-06-05:
iNav reports the GYRO/ACC as MPU6500, it works perfectly. My favourite iNav board.

DzikuVx, 2017-06-06:
> Sensor status: GYRO=OK, ACC=OK, MAG=OK, BARO=OK

Like @stronnag wrote: everything is OK since MPU9250 is 2 devices in 1 package

closed 2017-06-06 by DzikuVx