iNavFlight/inav #1767: Feature Request: Add HDOP to OSD

moggiex, 2017-06-05:
Howdy,

Is it possible to add HDOP as an option to display on the OSD please?

Reason: Might not have a low enough value when arming and would be able to see that, delay for a minute or two until the HDOP is a more acceptable value before taking off.

Matt

fiam, 2017-09-15:
@moggiex Not sure how to implement this one. Should we add a new OSD element for the HDOP or a setting the `OSD_GPS_SATS` which displays HDOP besides the satellite number?

moggiex, 2017-09-16:
Good question @fiam 

In an ideal world it would be a separate value to be shown on the OSD. Personally I would always have it showing under the sat count on my screens, Matt

fiam, 2017-09-17:
@moggiex How does this one look? It's just displaying the integer part of the HDOP to save some space. Do you think adding decimals is worth it?
[image]

moggiex, 2017-09-26:
Just seen your message, yes adding one decimal place is the "norm", Matt

fiam, 2017-09-26:
@moggiex We'll add a decimal place once we merge fixed width number formatting. I'll reopen this issue to avoid forgetting it.

fiam, 2017-10-03:
HDOP now uses one decimal, so this is now fixed.

PR #2168 (2017-09-26): Add support for showing GPS HDOP in the OSD
Requires https://github.com/iNavFlight/inav-configurator/pull/252
from the configurator side.

Fixes #1767
commit 043a67d2 (2017-09-26): Add support for showing GPS HDOP in the OSD

Requires https://github.com/iNavFlight/inav-configurator/pull/252
from the configurator side.

Fixes #1767
commit dc893a79 (2017-10-03): Use osdFormatCentiNumber() for HDOP

This allows showing one decimal for values < 10 without using
3 spaces.

Fixes #1767

(this time for real)
closed 2017-10-03 by fiam