iNavFlight/inav #1768: RSSI (FrSky XM+ via Channel 16)

plogibear, 2017-06-05:
Hi,

INAV somehow cannot detect the RSSI from Aux 14 w/c is outputted by the FrSky XM+ on channel 16. It does work on Betaflight.

Cheers.

isilcala, 2017-06-06:
I'm using XM+ with iNav without a problem. Shouldn't it be the aux 12?

digitalentity, 2017-06-06:
Channels 1-4 are flight channels (AETR), channel 5 is Aux 1 - therefore channel 16 is Aux 12. 
RSSI on XM+ works well on my setups.

Schalonsus, 2017-06-06:
@plogibear , did you buy from BG? Then you have to flash the rssi firmware to the xm+...

plogibear, 2017-06-06:
Yes as a matter of fact I did buy it from Banggood. Thanks for the tip @Schalonsus . I almost threw it out the window.

DzikuVx, 2017-06-07:
@plogibear if you don't mind, I will close this ticket since it's not INAV firmware related

closed 2017-06-07 by DzikuVx