iNavFlight/inav #1771: Can't get servo output...........

cdog4twenty, 2017-06-06:
I am using a colibri race FC, I am all green on the setup screen. I have connected the servo ground and signal to outputs 3&4 on the flight controller. I am powering the servos and fc from the same bec, but the servos are not feed power from and through the FC. I cannot for the life of me get any servo movement except a jitter when plugging in. I have tested the servos throuought and they work fine. Im lost. I have also played with the channels and servos in the servo tab and im still lost......... The esc connected to output 1 does work. Do I need something special to drive the servos?

moggiex, 2017-06-08:
Howdy,

A couple of daft things to check

- Do the servos actually work? (are you _Really_ sure?)
- Has iNav been set to be a flying wing or fixed wing model on the configuration tab?
- Is the board armed?
- Is your receiver inputs working correctly?
- Have you plugged the servo into the correct PWM outputs? (PWM 3 & PWM 4 for a flying wing)
- On the configuration tab, do you have the servo refresh rate set to 50Hz

Hope those pointers help, Matt

sambas, 2017-06-08:
target doesn't have any Servo outputs mapped
https://github.com/iNavFlight/inav/blob/master/src/main/target/COLIBRI_RACE/target.c

digitalentity, 2017-06-08:
Servos are not supported on COLIBRI RACE target

cdog4twenty, 2017-06-11:
Well that is terrible news. I have a few I was hoping to use. Can the situation be fixed? The colibri race has four outputs on the FC to connect to.

digitalentity, 2017-06-11:
It may be fixed along with timer cleanup https://github.com/iNavFlight/inav/pull/1591

cdog4twenty, 2017-06-12:
Would I just re-install the latest inav firmware for colibri race to see if it works?

digitalentity, 2017-06-25:
PR in place. Closing

closed 2017-06-25 by digitalentity

S-T-A, 2018-05-07:
i have a TBS colibri v1.2 on i nav 1.9.1 with the wing preset only one servo works. With servo output to aux selected the other servo works on my arm switch mixers don't work can i get help to het this working???