iNavFlight/inav #1772: Inav 1.7 Poshold issue

meban, 2017-06-07:
Hi .I am using zmr250 kit, ublox m8n gps and spracingf3 controller. Quad works perfectly when I activate poshold while I do not move in the air. But quad can not fix when I activate poshold while moving. It starts to round in the air. Going down is getting worse. What is the reason of this?

DzikuVx, 2017-06-07:
Magnetometer. Probably too close to power lines. While moving, more current
goes through them, so it makes stronger magnetic field and it is getting to
magnetometer

meban, 2017-06-07:
Thanks for the answer. However, I experienced the same problem on a beach very far from the city.

digitalentity, 2017-06-07:
@meban DzikuVx is not talking about high-voltage power lines. Wires that go from your battery to ESCs and motors transfer significant current when quad is flying. If the compass sensor is close to motors, ESCs, power supply lines, battery itself, buzzer - readings will be affected while copter is flying. You likely won't notice these issues on the bench.

meban, 2017-06-07:
You might be right. I use pcb for electricity distribution and on the controller. The problem may be due to this. Thank you.

parnz, 2017-06-08:
If your spracingf3 have onboard magetometer, Make sure you are not using it.
Use the one that built-in M8N instead.
Move M8N far away from power line maybe like [image]

meban, 2017-06-08:
I pulled the magnetometer over the controller. I put it on the GPS. Poshold is nice.But in Poshold mode I move the quad in any direction. When leaves the stick, quad starts to  rounding. Sometimes it corrects, sometimes it starts to turn worse. Is it still affected by the magnetic field?

DzikuVx, 2017-06-08:
Please record some Blackbox logs, hard to guess without any hard data

DzikuVx, 2017-07-04:
SInce there is no new data for last 3 weeks and most probably cause is magnetometer too close to power lines, I'm closing this ticket. If any new data appears, feel free to reopen. Thanks

closed 2017-07-04 by DzikuVx

enchantedforest11, 2019-04-12:
> If your spracingf3 have onboard magetometer, Make sure you are not using it.
> Use the one that built-in M8N instead.
> Move M8N far away from power line maybe like this: [image]

Hey what is the frame you are using here? Would like to check it out more. Thanks