iNavFlight/inav #1773: Docs for developer

cabinw, 2017-06-07:
Hi all
I'm a developer also love to fly FPV. I want to contribute to the wonderful project but it's really hard to get into such massive codes. So is it possible to have some docs for devs to get into the project quicker?

I think there are lots of developers like who wants to do something but we don't know how to start.

GTorreil, 2017-06-08:
docs/development?

parnz, 2017-06-08:
Following [Building in Eclipse.md](https://github.com/iNavFlight/inav/blob/master/docs/development/Building%20in%20Eclipse.md)
i have never been successful compile using Eclipse in Windows.

GTorreil, 2017-06-08:
If you have to use Windows, I believe the easiest way is to set up Ubuntu in a VM, share the folder with it and compile the source there...

DzikuVx, 2017-06-08:
In case of Windows 10, it is possible to set it up with `Bash on Ubuntu on Windows`. Works pretty nice. Also Cygwin option is OK, was using it before

cabinw, 2017-06-08:
@GTorreil Yes, but the current docs is mainly for building issues. I mean logicly.
BTW I'm using Mac

GTorreil, 2017-06-08:
@cabinw  that's right
good point you're using a mac, building is really easy on OSX :+1:

cabinw, 2017-06-08:
@GTorreil got any ideas about where to start reading the code?
My plan is to build the code first and then read navigation code

giacomo892, 2017-06-08:
I agrees there should be some logical maps to explain better how code is integrated and the logic beside it so new developers can contribute better and follow guidelines easily :)

DzikuVx, 2017-06-09:
There is one thing that developers do not like to do: write technical docs.
I'm affraid no such a documentation is available and honestly, I do not think it will ever be created :)

Anyhow, if you do not mind, I will close this ticket, since there is not much to do on the firmware side. @cabinw in case of questions, you are welcome to join team Slack channel to which I will gladly invite you if you want

closed 2017-06-09 by DzikuVx