iNavFlight/inav #1774: Feature request: FrSky and RSSI

RCnator, 2017-06-07:
Hi! I would like to be able to connect RSSI from my L9R and X8R receivers directly to my F4 Omnibus pro boards. There are plans of "Software PWM based RSSI" in Betaflight that would solve this,  [https://github.com/betaflight/betaflight/pull/2292](url), is that something we might get with iNav?

iconicj, 2017-06-08:
The L9R doesn't output pwm. It outputs an analog voltage of 0-3.3v, and works fine as is on the rssi input on the board.

moggiex, 2017-06-08:
@RCnator you can do this already. 

There is most likely a tiny pad on the bottom of the board that you solder to and then connect the RSSI output from the L9R or X8R to.

**PS.** It's an absolute bawbag trying to solder onto that pad. I used a spare dupont wire and then used a dab of dirty hot glue to keep it in place.

Below is an image of the Omnibus Pro V3 board, it's in a similar pace on the V2 board as well,

Matt 

[https://www.unmannedtechshop.co.uk/product_images/uploaded_images/omnibus-f4-pro-bottom.jpg](https://www.unmannedtechshop.co.uk/product_images/uploaded_images/omnibus-f4-pro-bottom.jpg)

moggiex, 2017-06-08:
PS. Remember to enabled ADC RSSI on the configuration tab in iNav the configurator too.

RCnator, 2017-06-08:
How can you all be so sure? Have you tested this yourselves, do you just assume this or have I missed something? Please see posts 1488 and 1489 in this thread about the F4 Omnibus pro, [https://www.rcgroups.com/forums/showthread.php?2801694-Omnibus-f4-pro/page60&perpage=25](url)

moggiex, 2017-06-08:
@RCnator I have 5 of the V2 boards with iNav on so "I'm pretty sure" :)

RCnator, 2017-06-08:
Well that is positive news then. What iNav version are you on? Due to the posts from Teralift on the F4 Omnibus pro thread on rcgroups I thought this was currently impossible. It was something about the ADC sensor already being used for current instead etc.

DzikuVx, 2017-06-09:
ADC on Omnibus F4 works in INAV. That means, both RSSI and analog Current Sensor are operational. 
Betaflight uses slightly different targets than INAV and indeed, some time ago, RSSI input was not working there.

closed 2017-06-09 by DzikuVx