iNavFlight/inav #1776: Problem with horizon drift.

marbalon, 2017-06-08:
My friend use iNav 1.7.1 in a 3.5M glider. When he make a circles in thermals he get a huge drift in angles.  

Situation is always the same. When he hold roll for a long time IMU roll angle calculation have gets huge drift and think that it is leveled....

The best if you see that movie:
OSD time:
23:00 - 24:08 - circles with roll stick
24:09 - 24:39 - circles without roll input, IMU "think" that plane is leveled
24:40  - roll stick in opposite side and big horizon error

Like you see on the video this is without motor so it is not because of motor noisy... The same situation is for pitch, sometimes he hold Pitch down a little when make a circles and when release model stayed pitch up.

Is there any parameters to tune to avoid this situation ? Filters, ACC weight in calculations ?

Sensor is MPU6000 over SPI. 

Any suggestions?

oleost, 2017-06-08:
Share blackbox log and cli dump

wiesiekkr, 2017-06-08:
Below you will find the cli dump, while the video will be tomorrow :)
[dump1.7.1.txt](https://github.com/iNavFlight/inav/files/1062038/dump1.7.1.txt)

Redshifft, 2017-06-08:
From description it's radio drift or FC sensor drift or I term pumping up - With a little imagination you could test this on the bench quite easily, also test it in passthrough

wiesiekkr, 2017-06-08:
I flew between trees in acro mode and pass and everything was fine :)
This is not a drift radio signal

rb1205, 2017-06-09:
I had this issue myself after a crash. I partially fixed it by doing a new accelerometer calibration as explained [in the wiki](https://github.com/iNavFlight/inav/wiki/Sensor-calibration#accelerometer-calibration-steps). I verified in the CLI that the relevant parameters changed by a noticeable margin, and a second calibration confirmed them with a good approximation. I say partially because I keep on experiencing some drift that I didn't notice before, but at this point I'm not sure if it's due to software or hardware issues.

Did your friend experience a crash previous to the issue? Does anyone know if a MEMS accelerometer can change its characteristics (but keep working) following a moderate shock?

digitalentity, 2017-06-09:
We certainly have issues with accelerometers on our boards. Their characteristics change with temperature and we don't compensate for it. I've also seen accelerometers killed by moderate crashes.

wiesiekkr, 2017-06-09:
In my case, it is not drifting due to temperature. I never had this model of any accident that could affect my work ACC

digitalentity, 2017-06-09:
@wiesiekkr  Ok, this might be something https://github.com/iNavFlight/inav/pull/1481 is going to fix. It was not tested yet, so caution is advised.

wiesiekkr, 2017-06-09:
Promised log file:
https://drive.google.com/file/d/0B3GW08HpYYLJek1qTnJrUk52Tm8/view?usp=sharing

oleost, 2017-06-18:
@DigitalEntity had a chance to look at log?

digitalentity, 2017-06-18:
@oleost @wiesiekkr looks like the problem that is unsolvable without additional external reference (like GPS). In a proper balanced turn accelerometer reads 1G, but in a wrong direction. Basically, plane indeed thinks it's flying level, while it's in a balanced turn.

I'm going to further develop and test #1481 once I get home

Dronek, 2017-10-05:
> Situation is always the same. When he hold roll for a long time IMU roll angle calculation have gets huge drift and think that it is leveled....

I can confirm this drift on my wing as well. It's enough to do a constant roll turn for a few seconds to get a slanted AHI in the OSD in leveled flight. Turning the other direction for a few seconds gets AHI back to level. Meanwhile I calibrated the ACC several times with quite similar results in terms of values. 

Also, I don't remember having this issue with 1.6.x versions on identical hardware.

Will try to reproduce this behavior in a bench test.

digitalentity, 2017-10-18:
I'm moving this issue to 1.9. The problem related to OSD updates is fixed, the problem with making longish banked turns remains, but it will have to wait a bit.

wiesiekkr, 2019-06-24:
Hello,
something going on in the topic?
When I flying a glider it is really very annoying :(
Below is a movie from the flight.
INAV 2.2.0

BR
Wiesiek

[https://www.youtube.com/watch?v=nRl5xqnDpMQ](url)

marbalon, 2019-06-24:
Btw. Wiesiekkr recalibrated ACC, and still have a problem. It is quite danger because when you got failsafe when IMU goes creazy you have big chance to see your plain will make something not expectect...

lgojniczek, 2019-06-24:
I have the same issue as @wiesiekkr has, and it is very annoying too.

wiesiekkr, 2019-07-06:
Hello,  Horizon Drift Fixed  - probably :)
I did only two flights, but everything looks promising.
 I entered the suggested value in cli :  imu_acc_ignore_rate  = 10
Below I'm sending you a link to the movie

https://www.youtube.com/watch?v=xfZFMtaLGnA&feature=youtu.be

BR 
Wiesiek

digitalentity, 2019-07-06:
Thanks for sharing your experience!
This was not a "fix", rather a "workaround", but at least it's something that works. 

I'll close this issue now.

a6j, 2022-06-22:
Hello. I haven't used INAV for two years. Now I use
the Matek F405-wse flight controller on two planes. I tried INAV 3.0, 4.1 and even release
candidate 5.0. And on all versions there is a problem with horizon drift. I really
like the ideology of INAV and I don't want to switch to Ardupilot. I tried to reduce
the sensitivity of imu_acc_ignore_rate = 7, but this gives only a small result
and does not solve the problem. It is currently impossible to fly on INAV. What are
your ideas?

MrD-RC, 2022-06-22:
The first thing you should do before trying to tune the IMU ACC ignore parameters is to make sure your build is as good as you can get physically and electrically. 

- Make sure there are no removable vibrations. For example imbalanced propellers, damaged motors etc. if the airframe has bad vibrations and you can fix them. You may need to soft-mount the flight controller. 
- Make sure there is no erroneous electrical noise. For example, if you can see noise in your video feed, that does not just effect the video. That can effect other components too, like the accelerometer. Clean the power with capacitors on the hardware that is causing the noise, the ESC for example. LC filters on the video equipment only hides the noise from them, not fixes it for everything else. In fact, decent VTxs with good filtering may even mask this issue too. 

Once the system is clean. The ignore rates can be tuned. 

To be honest though. I don't know why this issue was closed. Horizon drift still exists, and the band aid of the imu acc ignore parameters can only go so far. It ok to say that this problem does not have a simple solution. It is complex. But we should say it's fixed, when it isn't.

a6j, 2022-06-22:
I have been trying to overcome this problem for more than six months, but nothing is working. For example, on the Atomrc Dolphin plane, I managed to pick up a propeller group in such a way that this is the best thing I've had in the last ten years. There are no vibrations, with a cruise current of 3-6 A, the plane flies almost soundlessly. Even my friend noted it, which turned out very well. Even flying slowly without sudden maneuvers, the horizon is floating away. I'm disappointed that I can't fly planes right now because of INAV issues.

a6j, 2022-06-22:
I started using INAV from the first release. There were no problems with the horizon at the beginning, until a certain moment (release). I don't remember exactly, but somewhere in the second
branch, a similar problem with horizon drift began, which was surprisingly
quickly fixed. And starting from version 2.6, everything was broken again. People, let
's fix this misunderstanding! After all, this problem pushes the community away from
the project. A meme has already appeared on how to quickly get rid of the INAV horizon drift in
five minutes.

a6j, 2022-06-23:
I understand that everyone doesn't care about this problem? Is it really necessary to switch to the wretched Ardupilot...

Pairan, 2022-06-23:
I don't think people doesn't care ... I think it's more the thing that nobody has a good idea to solve it in the current constellation/development! I can't do it ... but I assume you can't either!

It would be wonderful if somebody took care about it and moves this issue out of the way FOR GOOD! ... 
And if it is as @breadoven assumes, that the design of INAV not offers any easy solution, then there must be a redesign. That happens all the time (here) although it's more the smaller things. But perhaps its time to rethink the old patterns ... 

Anybody up for this task? That would need somebody like Konstantin back then, who said: "I think I can do this!" ...

a6j, 2022-06-23:
Well, since no one can solve this problem, I understand that they don't want to. I have a plan: roll back to the release of five years ago where everything worked,
cut out all the wrappers and focus on the fundamental functions of INAV on which it was based. If this is not done, then INAV is dead.

MrD-RC, 2022-06-23:
@a6j I really don't think either if those things are true. Rolling back 5 years of work is really not going to happen. I'm sure that one of the first things to have happened when this issue was first discussed is that the difference between a "working" version and 1.7 would have been investigated. So rolling back is not the answer. 

Also, if after 5 years of having this issue, it will not kill iNav. It does need to be solved, but it is complex, and you are being melodramatic. If your plane is unflyable, as you have previously stated. I'm betting that there are excessive vibrations and electrical noise in the build. That is how large drifts occur. Smaller drifts from slow, constant turning etc, can be tuned out with the imu acc ignore parameters. 

I think @breadoven hit the nail on the head with "because nobody is smart enough to work out how to fix it". And that is not saying that people working on iNav aren't very smart. It says more to the complexity of the equations
involved, understanding of flight dynamics, fusion, and how iNav currently is coded.

breadoven, 2022-06-23:
You need to understand quaternion theory and all that nice IMU stuff and then program it. But understanding the quaternion stuff in itself probably requires more time and effort than most people are prepared to give up for free.

And isn't the big problem with Ardupilot related to the fact it simply won't fit on many boards due to its size, even after it's been split up into different craft types.

Pairan, 2022-06-23:
... dead? ... nah ... not at all! @MrD-RC did pretty much hit it !

Yet, it INAV doesn't suit your needs anymore then Ardu is surely not a bad choice. 

The problem here will remain until somebody will have an solution- thats how open source works, I guess!

But stating INAV to have died is a strange idea !

a6j, 2022-06-23:
Five years! Five years! And no result! Yes, INAV is dead in absentia, if nothing changes.

Pairan, 2022-06-23:
... not for me!

a6j, 2022-06-23:
You are not the universe.

a6j, 2022-06-23:
There is a solution. Cut out everything that is not necessary to leave the fundamental functions.

MrD-RC, 2022-06-23:
> Dude, are you inattentively reading my messages or my English is really bad? In previous messages, I emphasized and specifically pointed out that the plane does not have vibrations at all! I understand that all developers are looking for an excuse for their mistakes. You confirm once again that INAV is dead.
> 
> I can program, but in other languages in which INAV is built.

If you have no excessive vibrations or electrical noise, you will not have bad, unflyable horizon drift. Unless you have changed some settings which will make it worse.

Check:
```
acc_lpf_hz = 15
imu_acc_ignore_rate = 9
imu_acc_ignore_slope = 5
```
for starting point settings on 3.0 and above.

a6j, 2022-06-23:
Your example is the default settings that don't work.

a6j, 2022-06-23:
What's wrong?

MrD-RC, 2022-06-23:
Those settings work for well for most people. There can be some drift in long, slow turns. But nothing unflyable. And it can be minimised with tuning of the last 2 parameters. 

Do you have any DVR of this unflyable drift, so we can see how the AHI is behaving.

a6j, 2022-06-23:
Okay guys, I understand that since everything is fine with you, it means that it should be the same with everyone else. I assure you it's not like that.

a6j, 2022-06-23:
To summarize.
The horizon is floating away in INAV. What to do?
Switch to the wretched Ardupilot.
The problem is solved. You can close the topic.

MrD-RC, 2022-06-23:
@a6j how about posting that DVR?

We have been trying to help, but you have only complained. Please show us the unflyable horizon drift you are experiencing, so that we have a better idea of your problem.

digitalentity, 2022-06-23:
Sorry, @a6j is will not be able to answer for the next 7 days for not respecting our [code of conduct](https://github.com/iNavFlight/inav/blob/master/docs/policies/CODE_OF_CONDUCT.md)

shellixyz, 2022-06-23:
Horizon drift is a thing... but if it is to the point it makes your aircraft unflyable there is an issue with your setup. The horizon should only drift when there are accelerations and it is usually a few degrees. It should definitely not drift more than that and certainly not in straight lines.

Personally I have given up fixing that issue because it would require a lot of work and not everybody agrees on what should be done to fix it.

digitalentity, 2022-06-23:
As noted a few times here it boils down to the capacity of the developers working on the project. I started working on the proper fix at some point, but realized that it would take redesigning the whole IMU code (and potentially breaking a lot of things in process). Unfortunately I didn't have (and still don't have) the capacity to take on that task :disappointed: 

From what I observed so far, the unmanageable and unflyable horizon drift is typically caused by damaged/unstable accelerometer or severe vibration issues on the aircraft - especially if the drift happens when flying in a straight line.

Pairan, 2022-06-23:
from an upcoming episode- nothing happened and the wing flies good! 

https://user-images.githubusercontent.com/8381714/175374451-13fb7c1a-bf48-48e5-b209-bf43cd7dbc48.mov

shellixyz, 2022-06-23:
@Pairan That's quite a lot of drift...

Pairan, 2022-06-23:
I was somewhat surprised when those values showed up- but as I was in acro there was no Problem. Hitting RTH at that point would Have been a bad idea ;)

it recovered some minutes later - but something had glitched there: osd stated me to be at 200km/h and at -99m alt 🤣

120km/h and -5m was the reality

rb1205, 2022-06-23:
@a6j you are the only one complaining by unflyable horizon drift. You're the only one saying that iNav is dead. You're the only one pretending people to fix stuff for you. And then you have the audacity to tell "you're not the universe" to someone else? LOL.

danarrib, 2022-06-23:
I'm flying INAV airplanes for years and I never had problems with horizon drift. 
Flying 50 km missions without any problem.
I'm not saying that it doesn't exists... Just saying it doesn't happen to me.

As a piece of advice, maybe you should try:
- Lowering the PID loop frequency... Really, there's no point to use more than 1kHz on a fixed wing aircraft.
- Remove vibrations. Check motor and prop balance.
- Try to place the battery near the flight controller, as the battery mass will help dampening some of the vibrations. Use straps to fix the battery firmly to the airframe. Also make sure the FC is firmly fixed to the airframe as well.
- Remove electrical noise. Add a capacitor to the power input of the flight controller. If there's other noisy electronics near the FC, make sure they're shielded. Check grounding to avoid ground loops. Twist signal cables of video, servos, ESC, GPS and receiver with a ground wire to shield them.

shellixyz, 2022-06-23:
It is odd that you don't see any @danarrib. The "normal horizon drift", the one which happens in turns is due to the fact that the accelation component of the acceleration vector measured by the accelerometer is not taken into consideration. The whole acceleration vector is taken as being the gravity vector. So if you turn (and don't have the acc_ignore stuff setup) the measured vector doesn't point in the same direction as the gravity vector, that gets integrated by the AHRS and results in the horizon drifting in the direction of the turn (so when you let go of the roll stick in angle it has a tendency to continue rolling in the same direction). 

The issue should manifest on every aircraft even with a perfect setup since it is an algorithmic issue (and indeed does at least on mines).

The reason the drift shouldn't grow big is that, unless the lateral acceleration goes beyond 1g, the angle difference between the measured acceleration vector compared to the gravity vector will never go above half of the roll angle so even in a long turn or when circling the horizon drift should stay well bellow half of the roll angle.

a6j, 2022-07-01:
С таким подходом мистер Константин проблема никогда не решиться, я разочарован.

Pairan, 2022-07-02:
Currently I found an easy way to reproduce the drift! Simply put the plane/wing into PosHold and remain there for some minutes. 

the initial rounds will fly as expected but after some times the wing (or plane) will start to throttle up more and overshoot which INAV tries to compensate with lower throttle ... yet that results in steep dives and climbs at high throttle ... after some time it's reaching a dangerous level.

Further the effect seem to has increased in Inav 4.x/5 compared to 2.x - I can't recall having it in 2.6.x

Perhaps it's the velocity controller, that winds up and my settings need some love. But that is only speculation.

I don't want to add pressure yet I want to show how easy we will get there. I can provide footage from several of these moments. The planes/wings usually are without any vibration and perform great. 

Latest demo with INAV 5.0 : https://youtu.be/JyUgGNOPvcM

MrD-RC, 2022-07-02:
That long, slow accumulation of drift has been known about for a while. That is the kind that can be negated for the most part by tuning the `imu_acc_ignore_rate` and `imu_acc_ignore_slope`. Usually drift of that kind does not lead to an unflyable model, and will level off when flying straight again.

Pairan, 2022-07-02:
I would love to find a good documentation for the values in the Wiki ! ... currently moved my values to:
`imu_acc_ignore_rate =10` to 12
`imu_acc_ignore_slope=5` ... 

5 was already set

synersignart, 2022-07-30:
Have anyone have a good setup yet for Multirotors?
my Hardware is 
STM32F411CEU6
MPU6050
BMP180
QMC583

Problem is Horizon Drift that wont recover in Angle Mode . also have notice after shaking the drone on the bench the Pitch and Roll in Setup takes long to recover to Level position

shota3527, 2022-09-18:
trying to fix with pr #8403

pasyn, 2022-09-23:
Having the same issues on a fresh multirotor install. It is a hexcopter that flew very well with betaflight so definitely not a hardware issue in my case. I believe I am experiencing the same bug as above. The strange part is that the horizon display will drift so that it is at aprox. 45 deg. when I am holding the copter stable. In angle mode, it will try to level out with the horizon causing it to fly to one side.

> I would love to find a good documentation for the values in the Wiki ! ... currently moved my INAV 5 values to: `imu_acc_ignore_rate =10` to 12 `imu_acc_ignore_slope=5` ...
> 
> 5 was already set
Did this help at all?

> trying to fix with pr #8403
Has this pr been tested on a multirotor outside of the sim?

I am happy to test and provide logs.

closed 2023-03-06 by DzikuVx