iNavFlight/inav #1777: Servo slow react on movement of stakes

zvonimirtrg, 2017-06-08:
I installed iNav on naze rev5 on fixed wing and when i select angle mode servo very slow correction direction model and somehow all  commands have a delay. Servos are analog. Who knows what the problem I do not have gps connected to naze.

zvonimirtrg, 2017-06-08:
Can you send me Tuning tab for fixed wing

digitalentity, 2017-06-08:
It's different for all setups. Good starting point is a preset for an airplane of your size.

zvonimirtrg, 2017-06-08:
How to turn it on autotune?

digitalentity, 2017-06-08:
Please consider reading the docs https://github.com/iNavFlight/inav/blob/master/docs/Autotune%20-%20fixedwing.md. Also make sure you have read the wiki getting started pages https://github.com/iNavFlight/inav/wiki

For further support that please use RCG thread https://www.rcgroups.com/forums/showthread.php?2495732-Cleanflight-iNav-%28navigation-rewrite%29-project

zvonimirtrg, 2017-06-08:
Can you  send me dump file for the fixed wing, wing spam 1300mm

digitalentity, 2017-06-08:
Every fixed wing build is different. My dump won't help you if you don't understand how it works.

moggiex, 2017-06-08:
@zvonimirtrg try the 600mm wing preset, that should get you started. I've hardly tweaked any of my wings and they've flown fine with that preset (I can hear @digitalentity crying to himself).

Also as @digitalentity suggests above, 10 minutes reading the docs will definitely help, Matt

zvonimirtrg, 2017-06-08:
OK thanks

DzikuVx, 2017-06-09:
Since this is more of a support ticket, I'm closing it

closed 2017-06-09 by DzikuVx