iNavFlight/inav #1781: Target Request-Matek F405-OSD

SilentZen, 2017-06-10:
http://www.mateksys.com/?portfolio=f405-osd

Tested it with betaflight had 5 usable uarts, neat attachement to power distribution board which has current sensor (FCHUB6s), OSD, SDCard Slot.

Looks like a good replacement for omnibus.

digitalentity, 2017-06-10:
It's already in progress https://github.com/iNavFlight/inav/pull/1716

MaxFpv, 2017-06-24:
When will the release be available?

digitalentity, 2017-06-24:
@MaxFpv quite soon!

JulianGoesPro, 2017-06-29:
whats the benefit over the Omnibus?

SilentZen, 2017-06-30:
In my case the matek powers but omnibus does not, but seriously the Dev gets pinged when comments are made here and what makes one target better than another it's just a time waste.

teckel12, 2017-07-13:
Odd that they didn't include a barometer, however.

digitalentity, 2017-07-15:
Done (less I2C/baro/compass), usable for a flying wing though.

stronnag, 2017-07-15:
Plans for I2C? 

There are a number of users (including me) who would like to use the F 405 AIO (I2C pads exposed) with iNav on multirotor.

digitalentity, 2017-07-15:
@stronnag my sample of Matek board don't have I2C pads (V1 of the board).
I'll make required changes - would be glad if you can test them.

stronnag, 2017-07-16:
Thank you. I'll be pleased to test the changes, I should have the board within a week or so.

digitalentity, 2017-07-16:
@stronnag then I'll suggest we close this and open a new issue "Add I2C support for Matek F405". 1.7.2 will be release w/o i2c support, but we'll have it added as soon as possible. What do you think?

stronnag, 2017-07-16:
Good idea. The obvious title may prevent multiple issues.

closed 2017-07-16 by stronnag

McGiverGim, 2017-07-18:
If I'm not wrong, there're not a board version with I2C. Some images around internet shows a version with I2C pads, but I think this was a first version that never was sold. The real board that you can get has a DAC pad in place of the I2C pads.

sammichael25, 2017-08-27:
Would this also work on the Matek F405-AIO

digitalentity, 2017-08-27:
@sammichael25 if CPU pin mapping is the same - it would. I don't have the AIO board, so this isn't tested. If you have the board - please give it a try and report back.

stronnag, 2017-08-27:
AIO baord is wokring since https://github.com/iNavFlight/inav/pull/1920

They are the same electrically, the AIO  exposes the I2C pins.

stronnag, 2017-08-27:
And, if you need it,  there is a working (flight tested) hex file at  http://seyrsnys.myzen.co.uk/inav_dev/ (from 2017-08-19). It will be updated once #2041 is tested / merged and #2028 is merged.

sammichael25, 2017-08-27:
Ok thanks...Also what pins do I connect the external baro and compass too..do they both go on the same I2C pad? And can sonar be added?

stronnag, 2017-08-27:
Both mag and baro to the marked SDA/SCL pins (by the SD card). Clearly marked on the vendors documentation. My favourite iNav board ATM.

sammichael25, 2017-08-27:
I did that but cli status says Mag Unavailable....the mag icon on top is also red

stronnag, 2017-08-27:
We really need more that "doesn't work" to help debug the issue. Can you please provide:

* Brief description of the external devices you are using
* A CLI `diff`
* The output of the CLI `bootlog` command

For comparison, there's a working `diff` at http://seyrsnys.myzen.co.uk/inav_dev/nav_f405.txt (BN880 M8N GPS+HMC5883 and BMP280), the bootlog looks like:
````
# bootlog
Time Evt            Description  Parameters
   2:  0          CONFIG_LOADED 
   2:  1       SYSTEM_INIT_DONE 
 502: 19   TIMER_CHANNEL_MAPPED  (1, 1, 0, 2)
 502: 19   TIMER_CHANNEL_MAPPED  (2, 2, 0, 2)
 502: 19   TIMER_CHANNEL_MAPPED  (3, 3, 0, 2)
 502: 19   TIMER_CHANNEL_MAPPED  (4, 4, 0, 2)
 502: 19   TIMER_CHANNEL_MAPPED  (5, 5, 0, 2)
 502: 19   TIMER_CHANNEL_MAPPED  (6, 6, 0, 2)
 502:  2          PWM_INIT_DONE 
 502:  3       EXTRA_BOOT_DELAY 
1204:  9         GYRO_DETECTION  (7, 0, 0, 0)
1795: 10          ACC_DETECTION  (8, 0, 0, 0)
1894: 11         BARO_DETECTION  (0, 0, 0, 0)
2022: 12          MAG_DETECTION  (0, 0, 0, 0)
2022:  4       SENSOR_INIT_DONE 
2523:  5          GPS_INIT_DONE 
2523:  7    TELEMETRY_INIT_DONE 
3524:  8           SYSTEM_READY 
````

sammichael25, 2017-08-27:
`# bootlog
Time Evt            Description  Parameters
   2:  0          CONFIG_LOADED 
   2:  1       SYSTEM_INIT_DONE 
 502: 19   TIMER_CHANNEL_MAPPED  (1, 0, 1, 3)
 502: 19   TIMER_CHANNEL_MAPPED  (2, 1, 1, 2)
 502: 19   TIMER_CHANNEL_MAPPED  (3, 2, 1, 2)
 502: 19   TIMER_CHANNEL_MAPPED  (4, 3, 1, 2)
 502: 19   TIMER_CHANNEL_MAPPED  (5, 4, 1, 2)
 502: 19   TIMER_CHANNEL_MAPPED  (6, 5, 1, 2)
 502:  2          PWM_INIT_DONE 
 502:  3       EXTRA_BOOT_DELAY 
1188:  9         GYRO_DETECTION  (7, 0, 0, 0)
1779: 10          ACC_DETECTION  (8, 0, 0, 0)
1779: 11         BARO_DETECTION  (0, 0, 0, 0)
1779: 12          MAG_DETECTION  (2, 0, 0, 0)
3438:  4       SENSOR_INIT_DONE 
3942:  5          GPS_INIT_DONE 
3943:  6     LEDSTRIP_INIT_DONE 
4944:  8           SYSTEM_READY `

`# diff

# version
# INAV/MATEKF405 1.7.3 Aug 19 2017 / 13:15:00 (99c1939d)

# resources

# mixer

# servo
servo 0 1000 2000 1500 100 8 
servo 1 1000 2000 1500 100 9 

# servo mix

# feature
feature VBAT
feature SERVO_TILT
feature SOFTSERIAL
feature GPS
feature CURRENT_METER
feature LED_STRIP

# beeper

# map

# name

# serial
serial 4 2 115200 38400 0 115200

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# master
set gyro_sync = ON
set gyro_sync_denom = 1
set gyro_hardware_lpf = 188HZ
set gyro_lpf_hz = 70
set gyro_notch1_hz = 170
set gyro_notch1_cutoff = 125
set gyro_notch2_hz = 85
set gyro_notch2_cutoff = 43
set acc_hardware = MPU6500
set acczero_x = 45
set acczero_y = -8
set acczero_z = -31
set accgain_y = 4105
set accgain_z = 4103
set align_mag = CW270FLIP
set mag_hardware = HMC5883
set magzero_x = -2312
set magzero_y = -194
set magzero_z = -2144
set baro_hardware = NONE
set serialrx_provider = IBUS
set min_throttle = 1040
set max_throttle = 2000
set motor_pwm_rate = 2000
set motor_pwm_protocol = MULTISHOT
set align_board_yaw = 2700
set current_meter_scale = 165
set gps_sbas_mode = AUTO
set nav_rth_alt_mode = FIXED
set nav_rth_altitude = 1500
set osd_rssi_pos = 3095
set osd_gps_sats_pos = 3424
set async_mode = GYRO

# profile
profile 1

set mc_p_pitch = 80
set mc_d_pitch = 18
set mc_p_roll = 80
set mc_d_roll = 18
set mc_p_yaw = 95
set dterm_notch_hz = 125
set dterm_notch_cutoff = 90
set rate_accel_limit_roll_pitch = 2400
set rate_accel_limit_yaw = 360
set heading_hold_rate_limit = 30
set roll_rate = 40
set pitch_rate = 40`

stronnag, 2017-08-27:
Thanks (excuse me reformatting a bit). If I apply your config to my machine, it finds the mag:
````
# status
System Uptime: 13 seconds
Voltage: 42 * 0.1V (1S battery - OK)
CPU Clock=168MHz, GYRO=MPU6500, ACC=MPU6500, MAG=HMC5883
Sensor status: GYRO=OK, ACC=OK, MAG=OK, BARO=NONE, RANGEFINDER=NONE, GPS=NONE
SD card: Manufacturer 0x74, 15793152kB, 04/2015, v2.0, 'USDU1'
Filesystem: Ready
Stack size: 2048, Stack address: 0x10010000
I2C Errors: 0, config size: 3316, max available config: 131072
ADC channel usage:
   BATTERY : configured = ADC 1, used = ADC 1
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 2, used = ADC 2
  AIRSPEED : configured = none, used = none
System load: 2, cycle time: 1003, PID rate: 997, RX rate: 49, System rate: 9
````
Is the mag detected in Betaflight (the latest dev βF hex supports I2C on this board) ? Are there any I2C errors?

sammichael25, 2017-08-27:
Ok it is detecting now. Im not sure what went wrong. My last issue is controlling the gimbal. Im not sure which pins are mapped to servo 0 & servo 1.

stronnag, 2017-08-27:
Currently, there is a single multi-rotor servos defined in the target (on S1). I have no idea if this works.

sammichael25, 2017-08-28:
Is there a command to move the servo output to another pin? With servo gimbal feature enabled I cant use S1 for motor 1 and I don't know what other pin I have to move the esc wire too.

digitalentity, 2017-08-28:
There is a hardware limitations to possible pin assignments - that's the reason why resource remapping won't be available in INAV any time soon - it will create more issues than it will solve.

When pin1 is used to servo all motor outputs are shifted - pin2 is motor1, pin3 is motor2 etc.

sammichael25, 2017-08-31:
I tried that but only 3/4 motors work. Motor 4 which was moved pin 5 doesn't spin. I even moved it to pin 6 and same result.

stronnag, 2017-09-02:
See https://github.com/iNavFlight/inav/issues/2047 

There has been a hex available since 2017-08-28, testing in a standard quad configuration and specifically with I2C via the USART3 pins would be much appreciated.