iNavFlight/inav #1782: iNAV 1.7.1  Fixed Wing RTH Issue.

EnglishTurbines, 2017-06-10:
Unable to get the FC to go into RTH....(Flying Wing preset used)

In the Modes setup, Motor ARM works fine. on Aux1

 Passthru is ok and Horizon selects ok. on Aux 2

But on RTH, this will not turn Blue inside the Configurator and the Mode does not show up in the live OSD either....on Aux 3?

Is this a Bug?

Thanks

scrapit85, 2017-06-10:
Tested RTH while out flying yesterday about 10 times all worked fine.

EnglishTurbines, 2017-06-10:
Don't doubt it, are you using the same Version though? 1.7.1..?....What setting could prevent RTH working correctly inside iNAV..?....Thanks..:)

oleost, 2017-06-11:
Rth poshold etc won't activate while on ground.

So it won't turn blue in configurator

Closing this as that is normal behavior.

closed 2017-06-11 by oleost