iNavFlight/inav #1783: iNAV 1.7.1.....RTH for Fixed Wing....Suggestion

EnglishTurbines, 2017-06-10:
It is most unusual to have to choose a preset altitude to make the model fly back at during a RTH scenario.

 In most cases, when flying at range with a winged aircraft, losing altitude is a very bad idea, because altitude can almost always be converted into distance. 
  Usual practice, is for the aircraft to return to the Home position at the _same_ altitude which RTH was initiated. Then Loiter, circling and gradually losing altitude to a predetermined lower level with no landing attempts.

A caveat to this would be to have a _Minimum_ height to do a RTH where the aircraft would _climb_ to if an unexpected RTH scenario is initiated.

 My suggestion is to have the above as a _default_ RTH setting on a Fixed Wing scenario as with most commercial Autopilots.
 
 Thoughts..?

oleost, 2017-06-11:
There are already a minimum rth height set default. And it will climb to that height before turning home.

Also regarding landing when coming home is simple because it's an Failsafe feature, if your plane is home and you are still unable to get contact with it, it's better to try and get it home, than it running out of juice 100 meters up in the air.

You have also the option to configure Failsafe how you want.

Don't understand the issue here?

digitalentity, 2017-06-11:
By default during RTH aircraft will control altitude by "AT_LEAST" scenario (see wiki for explanation - https://github.com/iNavFlight/inav/wiki/Navigation-modes#at-least-predefined-altitude-above-launch-point-nav_rth_at_least_alt). Looks almost like what you are expecting so second @oleost here - please clarify your request.

EnglishTurbines, 2017-06-11:
Thanks ditalentity...Actually, there is a RTH altitude scenario "CURRENT" which is exactly as I described, just turn around and fly home at the current altitude, perfect for Fixed Wing and my needs. There are a few options it's true. Im noob to iNAV, but its transition into Fixed Wing with Autopilot is going to be a game changer for sure.
  I was expecting RTH Mode to select/turn Blue inside the configurator even if it won't work until after the model is physically launched and detects GPS movement / speed...
  Im more used to APM, but iNAV is way easier to use. I just hope it's doesn't become overly complex because that's the problem with APM....too clever for its own good.

 Thank you to all who helped me.....:)

closed 2017-06-11 by oleost