iNavFlight/inav #1784: iNAV Ver 1.7.1....Home direction Pointer Arrow.

EnglishTurbines, 2017-06-11:
A small bug, inside iNAV the OSD layout shows the Home Pointer Arrow as a numeric value instead of an Arrow Pointer.

 The Distance from home value also appears to have the Home Pointer Arrow stuck on the end of it for some reason?  Only inside iNAV though. Anyway, all's good now....:)
[image]

digitalentity, 2017-07-09:
Not sure what bug are you decribing - if on a real board everything it good and it's displayed incorrect only inside INAV Configurator - this bug should be reported there.

DzikuVx, 2017-07-28:
From my latest tests, home arrow works good. Perhaps you had old font uploaded? I'm closing this for now, if any new data appears, feel free to reopen

closed 2017-07-28 by DzikuVx