iNavFlight/inav #1785: nav poshold issue

94onur94, 2017-06-11:
I made a short flight and tried poshold, return to home functions. Drone stays in right position but when it stays, it is wobbling. My modes are angle, nav poshold and nav althold. What is the problem, why is it wobbling?

https://youtu.be/zG-CxxyKz_k

F450 Frame
Sp racing f3 deluxe controller
Dji 2212/920kv motor
1045 propeller
Ublox Neo-M8N GPS
Flysky fs-i6 transmitter / fs-ia6b receiver
3s 5200mah lipo

miradarya, 2017-06-11:
In think that's too high nav_mc_vel_xy_d, causing D term oscillation in the navigation velocity controller. I had a similar but less pronounced effect with my 9 inch quad, and reducing nav_mc_vel_xy_d to 25 solved it completely for me. In my opinion the default value is much higher than necessary to achieve good velocity control.

94onur94, 2017-06-12:
I reduced nav_mc_vel_xy_d to 30, it is better now. But it rounds around gps hold point. https://youtu.be/27eFvvXnatQ

giacomo892, 2017-06-12:
Are you sure that is a GPS hold related pid issue? I'm running an F450 and it is perfectly stable with stock settings. Also the battery is like yours.

I think that your Base/Acro PID's are completely untuned. Try to higher the P and the D gain on them and try the machine without GPS modes first.

94onur94, 2017-06-12:
I'm using default pid settings on INAV 1.7.1. Okey I will tune Base PID settings.

giacomo892, 2017-06-12:
On mine F450 i ended up with pitch/roll D to 45 and P to 60

vic2k, 2017-06-12:
Have the same problem. 
At no wind everything is ok, but on wind the kwad is wobbling as on video above.
F450
naze32 rev5
2212 motors
1045 props
3000 mAh 30C
1,1 kg total weight

profile 1

set mc_p_pitch = 55
set mc_i_pitch = 30
set mc_d_pitch = 23
set mc_p_roll = 45
set mc_i_roll = 30
set mc_d_roll = 24
set mc_p_yaw = 85
set mc_i_yaw = 45
set mc_d_yaw = 0
set mc_p_level = 16
set mc_i_level = 15
set mc_d_level = 75
set fw_p_pitch = 20
set fw_i_pitch = 35
set fw_ff_pitch = 10
set fw_p_roll = 25
set fw_i_roll = 35
set fw_ff_roll = 10
set fw_p_yaw = 50
set fw_i_yaw = 45
set fw_ff_yaw = 0
set fw_p_level = 20
set fw_i_level = 15
set fw_d_level = 75
set max_angle_inclination_rll = 300
set max_angle_inclination_pit = 300
set dterm_lpf_hz = 40
set yaw_lpf_hz = 30
set dterm_setpoint_weight =  0.000
set fw_iterm_throw_limit = 165
set pidsum_limit = 500
set yaw_p_limit = 300
set iterm_ignore_threshold = 200
set yaw_iterm_ignore_threshold = 50
set rate_accel_limit_roll_pitch = 0
set rate_accel_limit_yaw = 10000
set nav_mc_pos_z_p = 50
set nav_mc_pos_z_i = 0
set nav_mc_pos_z_d = 0
set nav_mc_vel_z_p = 100
set nav_mc_vel_z_i = 50
set nav_mc_vel_z_d = 10
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 60
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 30
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set tpa_rate = 0
set tpa_breakpoint = 1500

miradarya, 2017-06-13:
Circling around hold point in bigger and bigger circles (aka toilet bowling) is usually caused by motor/esc wires running too close to the mag sensor, or mag sensor not well calibrated. Are you using the mag sensor on the flight controller or on the GPS module?

94onur94, 2017-06-13:
@miradarya my gps position is as shown in the picture. I calibrated once but maybe mag sensor not well calibrated. I'm using mag sensor on the GPS module.

[image]

94onur94, 2017-06-14:
Today, I made PID tuning and now it's flying perfectly, there is no wobbling. But when I activate poshold, it rounds around hold point.

digitalentity, 2017-06-14:
Most likely mag alignment is not correct. On most GPS modules compass is placed on the bottom part of PCB and requires *_FLIP alignment.

DzikuVx, 2017-06-14:
> But when I activate poshold, it rounds around hold point

@94onur94 circling around target position is in 90% of cases due to magnetometer problems. Try calibrating again. But this not always really help. Sometimes they just broken

94onur94, 2017-06-15:
I calibreted magnetometer and this time it is better. It is hanging in the air. 

https://youtu.be/Kix6qILmPxQ
https://youtu.be/oSe90y_fRYc

giacomo892, 2017-06-15:
Is maybe the GPS stick too long and it is vibrating too much disruption magnetometer?

94onur94, 2017-06-15:
@giacomo892 in most drone video has long stick and they are hanging perfectly with m8n gps. What can I do for magnetometer? How can I decrease vibration?

giacomo892, 2017-06-15:
My GPS rod is about 10cm. Yours seems much longer than that from the video.

94onur94, 2017-06-16:
This is my current settings https://pastebin.com/i0Vyhrse. Poshold round problem continues.

digitalentity, 2017-06-16:
Toilet-bowling is always related to compass. It's either incorrect alignment, flawed calibration or interference. Please verify that when you point your copter to North, East, South, West heading reads 0, 90, 180 and 270 deg respectively. The value shouldn't change more than 5 deg when you tilt the copter up to 45 deg (pitch and roll)

DzikuVx, 2017-07-02:
Since there is no  new data, I'm closing this ticket. If new data appears, feel free to reopen

closed 2017-07-02 by DzikuVx

HoVietHien, 2024-05-23:
Hi @onur94 , Can you share me your PID, please