iNavFlight/inav #1786: Improving  flight modes monitoring via Smartport telemetry

kahira, 2017-06-11:
Presently information  about flight modes monitoring  via Smartport is  (only) found  here :

https://github.com/cleanflight/cleanflight/blob/master/docs/Telemetry.md

Tmp1 :
    1 is GPS Hold, 2 is GPS Home, 4 is Headfree
    1 is mag enabled, 2 is baro enabled, 4 is sonar enabled
    1 is angle, 2 is horizon, 4 is passthrough
    1 is ok to arm, 2 is arming is prevented, 4 is armed

Tmp2 : GPS lock status, Number is sent as 1234, the numbers are aditives :
    1 is GPS Fix, 2 is GPS Home fix
    not used
    not used
    number of sats

That is :   **as of the 27** in INAV 1.7.1 defined flight modes/submodes actually **only 6** are available
-ARM
-ANGLE
-HORIZON
-PASSTHRU
-RATE ( by exclusion logic)
-NAV RTH

Some catch up here  would be helpful . I suggest  adding   the following modes :

-TURN ASSIST
-HEADING HOLD
-BLACKBOX
-HOME RESET
-AUTOTUNE
-SERVO AUTOTRIM
-NAV LAUNCH

DzikuVx, 2017-09-05:
If this will be added, new S.Port message should be added without changing existing Temp1. SmartPort is very flexible protocol of new frames can be added without problem, just like INAV did with frame 420 that report disnce to home

digitalentity, 2017-10-13:
This needs to be discussed further. @teckel12 you may want to check if this is relevant to LuaTelemetry

digitalentity, 2017-10-18:
I believe we can close this issue. S.Port was massively reworked in 1.8.

closed 2017-10-18 by digitalentity

jaymal66, 2017-10-18:
Just for the record 
"the  multiple smartport telemetry improvements"  heralded with 1.7.3 include in fact  only 2 of the  7 suggested unreported  flying modes  ,..... Failsafe and later hurriedly added  Home Reset
Reason good enough to close the issue ?......I beg to disagree !

fiam, 2017-10-18:
@jaymal66 You don't need telemetry to tell you if modes without preconditions have engaged because you already know that in the ground with a properly configured radio. Telemetry is useful for modes that might not activate when they're requested because some preconditions failed (e.g. STAB needs accelerometer, RTH needs a home position, etc...)

jaymal66, 2017-10-18:
@fiam What is your Information Theory  background ?
All modes with or without precondition need to be confirmed because receiving no information  on the status of the mode  cannot be  interpreted as a confirmation that a status is indeed activated ....  even less so by  a "properly configured radio ".. as you candidly assert.
Why ?  because the lack of information could be very much due to other causes
..elementary watson !

fiam, 2017-10-18:
@jaymal66 If you have a switch configured to trigger AUTOTRIM and you flip it, AUTOTRIM is enabled. If it doesn't, your RC link is not working and you're in FS mode so you're not getting telemetry back anyway. There are no other causes. Same thing applies to all other non-conditional modes.

Calling that information theory feels like glorification to me. I'd say it's common sense.

jaymal66, 2017-10-18:
@fiam Thank you ...w're entering  Wonderland
'When I use a word,' Humpty Dumpty said, in rather a scornful tone, 'it means just what I choose it to mean — neither more nor less.'

digitalentity, 2017-10-18:
@jaymal66 It's more important to have modes with pre-conditions to be confirmed. There are modes that don't have pre-conditions - you know they should be active when you flip the switch on the radio. 

If they don't activate - you have a problem with your link and you have `FAILSAFE_MODE` reporting to inform you about that.

The pre-conditioned modes may not activate when you flip the switch on the radio - and you actually need telemetry to be made aware of that fact.

From my view S.Port flight mode reporting is serving a purpose of confirming flight modes that need confirming. This issue is closed.

jaymal66, 2017-10-18:
@digitalentity With you're  "you know they should be active when you flip the switch on the radio" the problem is what you suggest as  a fallback namely   "If they don't activate" 
How do know that " they don't activate"   if the the status is not reported ?

Beyond that it's worthy to note that SmartPort presently reports mainly modes with no precondition and neglect  those like TURN ASSIST  and NAV LAUNCH which are  preconditioned (or more aptly modifiers)
something which seems to  contradict your views
So do close  the issue since you're running the show but not because the matter  has been  thoroughly dealt with.

digitalentity, 2017-10-18:
You don't seem to understand that the only case when non preconditioned flight mode doesn't activate is when you don't have the link. In this case telemetry won't help you.

Re LAUNCH mode - it's not pre-conditioned by anything. If you flip the switch when machine is disarmed and then arm - it will activate. You'll notice a distinct beeping when it does. Same for TURN ASSIST - no preconditioning.

Also, in case you didn't notice - the issue is already closed.

digitalentity, 2017-10-18:
I believe this is the end of discussion regarding this issue. If anybody have a suggestion how flight more reporting can be improved (a practical one, not the "omg, it doesn't report this and that") - feel free to open a new issue.

I'm locking the issue so this last comment stays last.