iNavFlight/inav #1788: Running I2C error cause crash

Xe54, 2017-06-11:
Hi guys,
this weekend I had two really bad crashes with my brand new tbs discovery and iNav 1.7.1. It flown almost perfect, most of the time, but then out of nowhere it maked some weird flips and went to the ground. I gave it just another try with the same result. I flown few minutes without issues (RTH worked perfect out of the box), but then another flip and another crash... In both cases the were some thousands of I2C errors in the configurator to see and GPS and mag icons were both red. As result two broken props and some scratches on the motors. It's just a little bit frustrating...
And another thing: the idling problem I discribed in #1566 is much worse on this frame. If I'm trying to start on the slightly inclined ground three of four motors are hovering over the ground after few seconds without throttle. This can't be a desired behavior.

# bootlog
Time Evt            Description  Parameters
   8:  0          CONFIG_LOADED 
   8:  1       SYSTEM_INIT_DONE 
 511: 19   TIMER_CHANNEL_MAPPED  (1, 1, 0, 2)
 511: 19   TIMER_CHANNEL_MAPPED  (2, 2, 0, 2)
 511: 19   TIMER_CHANNEL_MAPPED  (3, 3, 0, 2)
 511: 19   TIMER_CHANNEL_MAPPED  (4, 4, 0, 2)
 511: 19   TIMER_CHANNEL_MAPPED  (5, 5, 0, 2)
 511: 19   TIMER_CHANNEL_MAPPED  (6, 6, 0, 2)
 511: 19   TIMER_CHANNEL_MAPPED  (7, 7, 0, 2)
 511: 19   TIMER_CHANNEL_MAPPED  (8, 8, 0, 2)
 511: 18  TIMER_CHANNEL_SKIPPED  (9, 8, 0, 3)
 511: 18  TIMER_CHANNEL_SKIPPED  (10, 8, 0, 3)
 511: 18  TIMER_CHANNEL_SKIPPED  (11, 8, 0, 3)
 511: 18  TIMER_CHANNEL_SKIPPED  (12, 8, 0, 3)
 511:  2          PWM_INIT_DONE 
 557:  9         GYRO_DETECTION  (2, 0, 0, 0)
 673: 10          ACC_DETECTION  (3, 0, 0, 0)
 688: 11         BARO_DETECTION  (3, 0, 0, 0)
 688: 12          MAG_DETECTION  (2, 0, 0, 0)
2352: 13  RANGEFINDER_DETECTION  (0, 0, 0, 0)
2352:  4       SENSOR_INIT_DONE 
2905:  5          GPS_INIT_DONE 
2905:  7    TELEMETRY_INIT_DONE 
2956:  8           SYSTEM_READY 

# 

# diff all

# version
# INAV/SPRACINGF3 1.7.1 Jun  3 2017 / 08:13:23 (48e1feff)

# reset configuration to default settings
defaults noreboot

# resources

# mixer

# servo

# servo mix

# feature
feature -RX_PPM
feature RX_SERIAL
feature MOTOR_STOP
feature SOFTSERIAL
feature GPS
feature TELEMETRY
feature CURRENT_METER
feature PWM_OUTPUT_ENABLE

# beeper

# map
map TAER1234

# serial
serial 0 32 115200 38400 0 115200
serial 1 2 115200 115200 0 115200
serial 2 64 115200 38400 0 115200
serial 30 1 57600 38400 0 115200

# led

# color

# mode_color

# aux
aux 0 1 5 900 1300
aux 1 2 5 1300 1700
aux 2 20 5 900 2100
aux 3 3 4 1300 2100
aux 4 9 4 1700 2100
aux 5 8 2 1500 2100

# adjrange
adjrange 0 0 6 925 1050 18 8
adjrange 1 0 6 1050 1175 19 8
adjrange 2 0 6 1175 1300 20 8
adjrange 3 0 6 1300 1425 15 8
adjrange 4 0 6 1425 1575 16 8
adjrange 5 0 6 1575 1700 17 8
adjrange 6 0 6 1700 1825 9 8
adjrange 7 0 6 1825 1950 10 8
adjrange 8 0 6 1950 2075 11 8
adjrange 9 1 6 900 1300 14 7
adjrange 10 1 6 1300 1700 13 7
adjrange 11 1 6 1700 2100 5 7

# rxrange

# master
set looptime = 1000
set gyro_sync = ON
set gyro_sync_denom = 1
set gyro_hardware_lpf = 188HZ
set gyro_lpf_hz = 70
set gyro_notch1_hz = 170
set gyro_notch1_cutoff = 125
set gyro_notch2_hz = 85
set gyro_notch2_cutoff = 43
set acc_hardware = MPU6050
set acczero_x = 115
set acczero_y = -2
set acczero_z = 4
set accgain_x = 4070
set accgain_y = 4074
set accgain_z = 4058
set align_mag = CW270FLIP
set mag_hardware = HMC5883
set mag_declination = 240
set magzero_x = 65
set magzero_y = -193
set magzero_z = -70
set baro_hardware = MS5611
set serialrx_provider = SBUS
set min_throttle = 1070
set max_throttle = 1980
set motor_pwm_rate = 2000
set motor_pwm_protocol = ONESHOT125
set failsafe_throttle = 1400
set auto_disarm_delay = 10
set gps_sbas_mode = AUTO
set gps_dyn_model = AIR_4G
set alt_hold_deadband = 10
set nav_use_midthr_for_althold = ON
set nav_manual_climb_rate = 500
set nav_rth_altitude = 1500

# profile
profile 1

set mc_p_pitch = 60
set mc_i_pitch = 10
set mc_d_pitch = 15
set mc_p_roll = 50
set mc_i_roll = 10
set mc_d_roll = 15
set mc_p_yaw = 80
set mc_i_yaw = 20
set max_angle_inclination_rll = 500
set max_angle_inclination_pit = 500
set dterm_notch_hz = 125
set dterm_notch_cutoff = 90
set rate_accel_limit_roll_pitch = 360
set rate_accel_limit_yaw = 0
set heading_hold_rate_limit = 30
set nav_mc_pos_xy_i = 180
set nav_mc_vel_xy_p = 255
set rc_expo = 60
set roll_rate = 40
set pitch_rate = 40

# profile
profile 2

# profile
profile 3

# restore original profile selection
profile 1

# save configuration
save

#

DzikuVx, 2017-06-11:
This suggest hardware issue. I flown MS5611 and HMC5883L today without any problems.
Bear in mind, that power on I2C connector on SPRF3 is 3.3V and it is possible that power drainage was too big for small onboard LDO. I had previously a case like that with one of my SPRF3

Xe54, 2017-06-11:
What is your suggestion? Another SPRF3 board? I hope you're right, the frame wouldn't survive much more crashes of that kind.

DzikuVx, 2017-06-12:
@Xe54 Try powering MS5611 and MHC5883L from 5V line. It should work fine. Or use external LDO that will give you stabilized 3,3V. LM7833 or other. But this is only a guess, nothing for sure

digitalentity, 2017-06-12:
Seems a hardware issue to me. Recently it's really hard to come by quality SPRacingF3 boards and HMC5x83 mags. Most of them are cloned/fake/crappy quality.

I wonder why nobody is producing GPS modules with MAG3110 chips or similar.

Xe54, 2017-06-18:
Another weekend and two crashes more...
@DzikuVx I tried out your suggestion. I got a new SPRF3 board and powered I2C bus from an external 5V bec directly. Nothing changed, after few minutes of flight in position hold mode came first crash. Then I tried alt hold mode with the same result. Here is the screenshot of the INAV window after second crash: 
[image] Any idea what this could be?

myvilli, 2017-06-18:
Hi,
I have removed the 883L chip, and check if I had less i2c errors: 11 errors on connection, then no more problems.
I can't have stabilise mode nor accro working in Inav 1.7, any guess? Flights mode are correctly configured.
Thanks.

diode121, 2017-06-21:
Are you using GPS module with Magnetometer?
If so, try adding pull up resistor on the SCL and SDA lines.

GTorreil, 2017-06-23:
I had the same problem before (M8N + mag module). I removed the magnetometer from the SPF3 and bought an external MAG3110 which I stick right on top of the frame. No more errors and great PH since then!

Xe54, 2017-06-23:
Thank you for your advices. I think I'll try it with another gps module.

Xe54, 2017-07-10:
You're right guys, this was a hardware issue. I changed the gps module and now it works fine.

closed 2017-07-10 by Xe54