iNavFlight/inav #1791: Mag sensor in BS-880 rarely detected at start up.

KiwiDavid, 2017-06-12:
Mag sensor in BS-880 (HMC5883) is rarely detected at start up using SPRacingF3 clone board with either iNav 1.6.1 or 1.7.1.
Appears to be a marginal timing issue as it will occasionally get detected and eventually allow arming but usually it fails and the red LED flashes rapidly.
Sensor is instantly detected following any "Save/Reboot" command so it's not a hardware related issue.
At a guess I would say the sensor needs a little longer to settle following power up so that it can respond to the micro when it's polling available hardware.
It's possible that the regulator used in the GPS/mag combo has a bypassed reference voltage (for lower noise) which can add further delays in start up.
Given that we all have to wait for a 3D fix when using GPS would a few hundred milliseconds extra settling time at startup really be a problem?

Regards,
David

KiwiDavid, 2017-06-13:
After reading through many other issues I believe I can confirm that I'm seeing the same issue as #727. 
I can also confirm that at around 24C the start up from main battery is totally reliable whereas at around 15C the start always fails.  A soft reboot via the USB is always reliable and this perhaps indicates the mag sensor has not settled (when starting from battery) by the time the micro checks for it.
A temperature dependence like this could give rise to seasonal errors after months of reliable operation.

Regards,
David

digitalentity, 2017-06-14:
Not much we can do about it. Increasing start-up delay more is probably not a very good idea - it's rather long already. However, you can try increasing the delay in `fc_init.c` here:
```
#if defined(GPS) || defined(MAG)
    if (!isMPUSoftReset()) {
        addBootlogEvent2(BOOT_EVENT_EXTRA_BOOT_DELAY, BOOT_EVENT_FLAGS_NONE);

        LED1_ON;
        LED0_OFF;

        for (int i = 0; i < 5; i++) {
            LED1_TOGGLE;
            LED0_TOGGLE;
            delay(100);
        }

        LED0_OFF;
        LED1_OFF;
    }
#endif
```

KiwiDavid, 2017-06-14:
Hi,
Thanks for getting back so quickly.
I thought it may be related to start up delay but I've had a scope on it and if I trigger off the 5V power coming up (ESCs) I'm seeing 1.85 seconds to the first I2C clocks.   Actually there's a single clock pulse at around 1 second)  That should be heaps of time for anything to settle so I doubt it's related to that. 
Was surprised to see the bus running at 400kHz (fast mode) but it shouldn't be a problem.  The temperature sensitivity is intriguing - about a 10 degree window between 100% reliable start ups and 100% no go.    The BS-880 GPS/mag combo takes 5V in and has an on-board 3.3V regulator for GPS and mag sensor.  Any supply above 3.6V should be fine but powering off 3.3V means the regulator is saturated and looks like a switch with no supply noise rejection.
Will do a bit more delving with a logic analyzer tomorrow.   Will also have a look at your code change but I may look in to adding a hardware reset to the FC.  Anyone know if I can safely just ground pin 7 on the micro and force a reset in the field after power is all settled?   Desperate times.....
Can't help but think the extreme temperature sensitivity holds a clue to something we're overlooking.

Regards to all,
David

KiwiDavid, 2017-06-15:
I can confirm that a brief grounding of pin 7 on the micro will take the device out of the mag hardware failure mode when cold and restore normal operation.  This is similar to the action of a "Save/Reboot" command. I'll do further tests at hopefully a lower temperature just to be sure.

Just getting in to it with the logic analyzer.  I can see the MPU6050 being almost constantly addressed with the other sensors accessed far less frequently.  Hoping I may be able to catch the difference between cold (fault) mode and normal warm mode.

KiwiDavid, 2017-06-28:
Had a couple of great runs using RTH and it behaved perfectly but initializing still very difficult.
Have recorded a few seconds of a bad start and a good start.  Bad start has no addressing of mag sensor anywhere whereas the good start has a very lengthy mag sensor start process and then periodic reads of the mag.

Bad start.
0.130935875,29,w (0x77),'172' (0xAC),Write,ACK
0.1309875,30,w (0x77),l (0x6C),Read,ACK
0.131010375,30,w (0x77),'132' (0x84),Read,NAK
0.13106825,31,w (0x77),'174' (0xAE),Write,ACK
0.131119875,32,w (0x77),[ (0x5B),Read,ACK
0.13114275,32,w (0x77),'11' (0x0B),Read,NAK       ****0.6 sec gap start ******
0.745652125,39,h (0x68),; (0x3B),Write,ACK          ****0.6 sec gap end ****** no mag sensor!
0.74570375,40,h (0x68),'1' (0x01),Read,ACK
0.745726625,40,h (0x68),\t (0x09),Read,ACK
0.7457495,40,h (0x68),'255' (0xFF),Read,ACK
0.745772375,40,h (0x68),'164' (0xA4),Read,ACK
0.74579525,40,h (0x68),'16' (0x10),Read,ACK
0.745818125,40,h (0x68),o (0x6F),Read,NAK

This shows the end of a baro read and then reverts back to a gyro/acc read - no mag read
Note the time gap after 0.13114275 seconds (over half a second) where the bus is locked by clock stretching (??) but the micro continues on.

Good start.
0.130933625,29,w (0x77),'172' (0xAC),Write,ACK
0.13098675,30,w (0x77),l (0x6C),Read,ACK
0.131009625,30,w (0x77),'132' (0x84),Read,NAK
0.131066125,31,w (0x77),'174' (0xAE),Write,ACK
0.13111775,32,w (0x77),[ (0x5B),Read,ACK
0.131140625,32,w (0x77),'11' (0x0B),Read,NAK
0.131227625,33,'30' (0x1E),\n (0x0A),Write,ACK
0.131279125,34,'30' (0x1E),H (0x48),Read,NAK
0.181604125,35,'30' (0x1E),'0' (0x00),Write,ACK
0.1816285,35,'30' (0x1E),'25' (0x19),Write,ACK
0.1816845,36,'30' (0x1E),'1' (0x01),Write,ACK
0.181708875,36,'30' (0x1E),'128' (0x80),Write,ACK

This shows the end of a baro read and the beginning of the mag read - no time gap or I2C bus being locked.  
I thought initially it was the mag sensor screwing up but in the bad start there is no addressing of the mag so it may be the baro is locking the clock line on the last NAK and the mag is never addressed and in the 0.6 seconds of no I2C bus the micro has flagged the mag as faulty. 

I have ordered a new GPS/mag combo but I don't believe the existing hardware is faulty - perhaps marginal on timing.  I don't think the present code deals with clock stretching so there may be a tendency to flag timing issues as faulty hardware.  Will progress on .....

Cheers,
David

digitalentity, 2017-06-28:
@KiwiDavid good finding! Please continue your experiments.

KiwiDavid, 2017-07-01:
I can find no evidence of clock stretching (with scope or logic analyzer) so my above assumptions are incorrect.  I can not think of any reason why the mag sensor is not at least addressed in a bad start unless directed by a code branch.  Even if the sensor was dodgy and couldn't reply the host would still talk to it and this would be seen on the bus.
Multiple bad starts all indicate a problem around 1311mS from the first -ve going I2C clock edge after power up.   This coincides with the end of the baro read.  There is then a 610mS delay before any further I2C bus activity occurs which is back to the gyro/acc sensor.  The mag sensor is effectively ignored and never again accessed leading to a hardware failure.

 My device fails below or around 20C and looking at the baro code there are at least two instances of temperature dependent lines of code.  One of them starts like this-

if (temp < 2000) { // temperature lower than 20degC
delt = temp - 2000;
delt = 5 * delt * delt;
.......

There is also a -15C code correction for very low temperatures.
I'm sure this is not a bug because people in cold areas would be frequently seeing problems but it is curious how the mag sensor is skipped in temperatures below 20C on my unit and a few others I've about.  I remain unconvinced that this is a hardware problem.
I'm not sure I can take this further as I'm not a C programmer.

Regards,
David

randyoo, 2017-10-14:
This issue is reproducible for me on an Omnibus clone (f4 v5pro) on 1.7.2 with a Beitian BN-880 GPS/compass. There's no blackbox capability on this FC due to lack of SD card slot, or I'd capture and share a log.

Engaging Pos Hold after a cold boot seems to always result in the aircraft flying away, but after a soft reboot is performed, PH works fine. I haven't yet observed an occasion where PH didn't work correctly after a soft reboot. Hard power cycle is hit or miss.

Even if the fault can be blamed on hardware, it still seems as if the software ought to react differently to a hardware failure of the mag, i.e. refuse to arm, or refuse to enter Position Hold mode, if the mag isn't working, or is giving erroneous data.

If desired, I can capture video of OSD (showing heading and GPS coordinates) during a flyaway event, or attempt to connect to the configurator software after a flyaway event, and then perform any CLI commands that might shed some light on the cause of the issue.

closed 2018-01-20 by DzikuVx