iNavFlight/inav #1792: NAZE32 working BUT need help with channel remap

Tony7575, 2017-06-12:
Hi, I am using an older board Naze32 rev.5, GPS works on UART 2, using up inpu 3 and 4.

SBUS not working because only 2 UART are available and 1 is used by USB.

Using PWM receiver, and channel 1 and 2 work like a charm, CH3 is supposed to be throttle but ch3 is taken by the GPS and Ch4 too, so how can I remap CH5 to THrottle and CH6 to Rudder?

I heard there is a MAP command in the CLI... how can I use this?

GTorreil, 2017-06-12:
MAP is used to map motor/esc outputs, **not** input channels in parallel PWM mode. 
You can use SoftSerial for your GPS  module, and the hardware UART for Sbus, it will work fine. 
Though, you can't SBus on a software serial port because of the high data rate, so keep it on UART2.

Tony7575, 2017-06-12:
HI can you point me to an explanation of how to setup software serial is it done in the configurator, I assume? It would be cool if I could just use channel 5 and 6 for throttle and rudder inputs... with a gui based remap.... interested in this softserial if I can learn about how to set it up.
Thanks,
Tony

digitalentity, 2017-06-12:
@GTorreil on the contrary. `map` command in CLI is used to specify input channel order.
`map AETR1234` corresponds to roll, pitch, throttle, yaw, aux1..4.

GTorreil, 2017-06-12:
you need to enable the **softserial** feature in the configuration tab. **Save and reboot**, then go to the **Ports tab** and you have two additionnal UARTs which are CPU Based. They can handle your GPS and any serial com with low bitrate. 
You're right about Map having Rx features, i had mised it. However, using parallel PWM Rx is not recommended as it eats up all your pins while limiting the number of channels you can get

GTorreil, 2017-06-12:
[image]
Take a look at this for the pinout ;)

GTorreil, 2017-06-12:
@digitalentity that's right, I got confused with mixer ^^

Tony7575, 2017-06-12:
Actually it all worked, I used throttle on input 5 and rudder on input 6 without issues at all, it  auto-assigned the channels to the correct function somehow in the order Ail-Ele-Thr-rud too bad now I only have 2 left for Aux channel, one for arming the plane , one for RTH is good enough for me.

DzikuVx, 2017-07-01:
Seems like problem solved, closing.

closed 2017-07-01 by DzikuVx