iNavFlight/inav #1793: Activating AltHold while inverted

digitalentity, 2017-06-12:
As reported on RCG (https://www.rcgroups.com/forums/showpost.php?p=37700755&postcount=14332)

>  Hi, I have a 220 size quad with 2300kv, 4s and 5x4.5x3 props, Omnibus F3. I fly line of sight, acro, practicing freestyle and FPV. I use iNav for poshold/althold and RTH as a safety measure.
They work fantastic when almost level, but when the quad is on his side or upside down and I activate one of the poshold/RTH modes, it first goes full power and than tries to level. It scared the hell out of me! It needed something like 30 meters horizontal and 10 meters altitude to recover. It stopped less than a meter above the ground.
Any ideas to improve this? I expected it first to rotate fast to level and than power up to hold altitude.

AltHold needs to be aware of inverted state.

KenImhof, 2017-07-04:
This feature is very much needed and I hope it can get fixed/implemented.

KenImhof, 2017-09-16:
Make some progress by adding a few lines of code.  If inverted reverse motors than reset home position at the top of the flip.  Sort video example that would cut-off during the RTH:
https://youtu.be/5oiWQhGGE78

oleost, 2017-09-17:
Looks very promising.

oleost, 2017-09-17:
Maybe do an pull request and take discussion from there?

KenImhof, 2017-09-17:
Submitted Pull Request #2170.

KenImhof, 2017-10-08:
Incorporated Rescue and Hard Deck modes in iNav3d.
Sort Demo Flight:
https://youtu.be/uNKoen8VKb4

Details and download:
https://github.com/KenImhof/iNav3d/releases

It would be great if we could get these features incorporated in the iNav base.

digitalentity, 2017-10-08:
Nice. Let's target this towards 1.9 along with `rcCommand` refactoring (https://github.com/iNavFlight/inav/pull/2171)

KenImhof, 2017-10-08:
Either 1.9 or 2.0 is a good target as I agree that the rcCommand refactoring needs to be done first.  Some of the 3d Hard Deck/Rescue modifications were to address 3d prop reversal compatibility issues with navigation modes.  The rcCommand refactoring should address these issues and should be solid before Hard Deck/Recuse modes are incorporated.

digitalentity, 2017-10-08:
Great! Once we get 1.8 out of the door I'll start the refactoring. Your assistance regarding 3D mode will be greatly appreciated :heart:

digitalentity, 2018-02-25:
`rcCommand` refactoring is stalled. So this is now blocked by #2171

closed 2023-10-27 by DzikuVx