iNavFlight/inav #1795: Lux FC - servos do not work

2ndalpha, 2017-06-12:
Based on this issue I might be only one of a few who is trying to run iNav on Lux: https://github.com/iNavFlight/inav/issues/1757

I do not get servos to work, they jump when plugging in the battery and then nothing.
They do not try to hold any position. I'm trying it in passthrough mode.

I tried the same steps/setup with Naze, worked as expected.

Might it be related to TODO-s here? https://github.com/iNavFlight/inav/blob/master/src/main/target/LUX_RACE/target.c

digitalentity, 2017-06-13:
Yes, servos are not implemented for LUX target. We'll probably fix it along with merging https://github.com/iNavFlight/inav/pull/1591

digitalentity, 2017-06-25:
PR in place. Closing

PR #1827 (2017-06-25): Allow servos for LUX_RACE and COLIBRI_RACE targets
Fixes #1795 and #1771

For airplane setup user will have S1 for motor and S2-S4 for three servo outputs.
closed 2017-06-25 by digitalentity