iNavFlight/inav #1797: inav pre flight navigation is safe..

locutus40, 2017-06-13:
i have a omnibus f3 board with this gps https://www.banggood.com/UBLOX-NEO-M8N-BN-880-Flight-Control-GPS-Module-Dual-Module-Compass-p-971082.html?rmmds=myorder
anyway i set it all up radio works gps says 11 sats and 3d fix mag and acel calibrated all checked off except 
navigation is safe... thats a red x..

also the modes i set up rth and head free also nav pos hold.. i click save but nothing turns blue when i function the switch... angle and beeper turn blue but not arm or any of the other things..

could this be related? and how do i get a green check for the navigation is safe?

locutus40@gmail.com

thanks guys btw im in doors next to a window 5 feet from it good signal red and blue lights on the gps

also it was all marked fine till i turned it on the next day i dont get it.. is there a way to disable navigation is safe so i can fly? a list of cli commands to disable the pre flight checks would be nice... each one that is

DzikuVx, 2017-06-13:
@locutus40 Screenshot would be very useful. Also result of CLI commands: `status` `bootlog` `tasks`

locutus40, 2017-06-13:
here is a screen notice the red x on the upper right.. 16 sats im indoors... everything else working..

not sure how to do cli commands im new to this stuff http://www.mediafire.com/view/uqhjj28iji8dbu4/Screenshot_2017-06-12_23.50.05.png

dont worry its safe dont know howt else to add links

DzikuVx, 2017-06-13:
Open CLI tab, copy-paste commands one by one and give back results they give you. Also one by one. Simple, right?

locutus40, 2017-06-13:
yea i can do that  but what do i put in so that you get what you need to help me? thats what im asking..

DzikuVx, 2017-06-13:
Let me quote myself:

>Also result of CLI commands: `status` `bootlog` `tasks`

locutus40, 2017-06-13:
ok let me type those in one by one see what it says
 give me a min i have to re connect my copter

locutus40, 2017-06-13:
Entering CLI Mode, type 'exit' to return, or 'help'

# status
System Uptime: 27 seconds
Voltage: 2 * 0.1V (1S battery - NOT PRESENT)
CPU Clock=72MHz, GYRO=MPU6000, ACC=MPU6000, BARO=BMP280, MAG=HMC5883
Sensor status: GYRO=OK, ACC=OK, MAG=OK, BARO=OK, SONAR=NONE, GPS=OK
SD card: Manufacturer 0x41, 3784704kB, 05/2015, v2.0, 'SD4GB'
Filesystem: Ready
Stack size: 2048, Stack address: 0x10002000
I2C Errors: 0, config size: 3304, max available config: 4096
System load: 2, cycle time: 2013, PID rate: 496, RX rate: 521, System rate: 9

#

locutus40, 2017-06-13:
# bootlog
Time Evt            Description  Parameters
   8:  0          CONFIG_LOADED 
   8:  1       SYSTEM_INIT_DONE 
 511: 19   TIMER_CHANNEL_MAPPED  (1, 1, 0, 2)
 511: 19   TIMER_CHANNEL_MAPPED  (2, 2, 0, 2)
 511: 19   TIMER_CHANNEL_MAPPED  (3, 3, 0, 2)
 511: 19   TIMER_CHANNEL_MAPPED  (4, 4, 0, 2)
 511: 18  TIMER_CHANNEL_SKIPPED  (5, 4, 0, 3)
 511: 18  TIMER_CHANNEL_SKIPPED  (6, 4, 0, 3)
 511:  2          PWM_INIT_DONE 
 521:  3       EXTRA_BOOT_DELAY 
1210:  9         GYRO_DETECTION  (6, 0, 0, 0)
1513: 10          ACC_DETECTION  (7, 0, 0, 0)
1533: 11         BARO_DETECTION  (4, 0, 0, 0)
1533: 12          MAG_DETECTION  (2, 0, 0, 0)
3196: 13  RANGEFINDER_DETECTION  (0, 0, 0, 0)
3196:  4       SENSOR_INIT_DONE 
3899:  5          GPS_INIT_DONE 
4904:  8           SYSTEM_READY 

#

locutus40, 2017-06-13:
# tasks
Task list         rate/hz  max/us  avg/us maxload avgload     total/ms
 0 -       SYSTEM       9       9       3    0.5%    0.5%         3
 1 -          PID     496     294     244   15.0%   12.6%     10199
 5 -           RX     514      64      54    3.7%    3.2%       423
 6 -       SERIAL      99    3840       5   38.5%    0.5%       131
 7 -      BATTERY      49      17       8    0.5%    0.5%        34
 8 -       BEEPER      99      20       8    0.6%    0.5%        62
 9 -          GPS      24     392      81    1.4%    0.6%       188
10 -      COMPASS       9     244     234    0.7%    0.7%       204
11 -         BARO      43      42      28    0.6%    0.6%       103
18 -          CMS      59      13       5    0.5%    0.5%        25
Task check function            17       8                        61
Total (excluding SERIAL)                    23.5%   19.7%

#

locutus40, 2017-06-13:
there you go any ideas? everything looks good to me all sensors working i dont get why it says  navigation is not safe :-(

parnz, 2017-06-13:
I also have this problem sometime with Navigation is Safe check fail.
I've to unplug-replug battery several times until I can ARM it.
Sometime I've all green mark but my quad not response to ARM switch. No clue.

locutus40, 2017-06-13:
even if i put my battery on it still dont clear that last check for navigations although i got 18 satalites now its been hooked up for 30 min see mine was all green at one point cept the mag cal and acel once i did those poof kicked gps out.. im stummed i got about 50 bucks into this little 130 frame now and id like to fly it

locutus40, 2017-06-13:
i think ive figured it out its with the inav.. i turned off my radio it clears everyhing to go..

turn it on then ir red checks the navigtation is safe.. hmm so i removed the nav position hold in modes.. and then just added nav alt hold or .. im gunna try and arm it..... but it seems nave position hold which is like dji's version of gps hold in 3d space does not work in inav... but the alititude hold does not sure why..

locutus40, 2017-06-13:
nope wont arm hmmm

locutus40, 2017-06-13:
ok i figured it out...parnz why your not arming and me not arming hook up your copter to inav turn on your transmitter.. then go into reciever where you see throttle.. see what it says at bare minimum mine was 1170

now type in in cli  type this in at the bottom get min_check it shoud give you a number mine was 1000.. so basially its like having your throttle turned up so if you arm your copter will fly off.. what i did was this..

i have a dx8 went in to servo setup  then travel and youll notice 100% and 100% throttle down it should highlight the 100% associated with the min throttle now at the bottoms theres a number 0-100 L or R stands for left of right.. watch the reciver screen on inav adjust this wiht your up or down or scroll so that it goes to like 140 % or less like 90% until you reach what the min check yours was set to below that mine was 1100 i set my throttle lower like 1091 and poof it arms.. dont use the trims... follow this guide..

https://www.youtube.com/watch?v=M0S3HTQihxs

he does other stuff.. so bascially mine woulndt arm besucz my thrttle valuve was waaay to high..

also the red check on gps thing i was talking about i figured it out you cant have it on gps aka your gps mode then arm it.. you have to turn off all modes aka you gps modes you can start in horizon and angle mode.. but no mode in gps so arm your copter.. then when the props are spinning you switch to gps mode or other gps modes.. it just cant start in them make sure to do this before you take off..

also if your motors spin alittle one spins the others dont.. i had to do this.. go into configurateion tab
then to the right esc motor features check  enable motor and esc to turn your motor pwm control on.. then go to  min throttle below that should be 1000 set it to 1070 thats the speed your props will spin at idle when you arm other wise they will twitch.. and one will spin and others wont dont mean that it wont fly just anooying...  next one down 1500.. then 2000 then 1000 for the other tabs..

now it should arm motors spin.. so before you fly set your modes.. make sure your throttle sticks in reciever go to the middle then watch it should say 1500 on yaw and rudder ale.. if they are off.. go in to your menu or even use the trims to get them all to say 1500 even the throttle one... or you can use the sub trim menu which im gunna do in a bit i used trims.. so sticks all center 1500 sticks max you can go into sub trims to get your throttle up to 2000 your yaw etc etc..

once i wrapped my mind around it it became simple.. see ive set up dji stuff and other models like 3dr this is by far more involved mission planner is similar with the sub trims but i forgot about it it finally rinnged and then i tried it on this and said wait a minute lol hope this helps you

locutus40, 2017-06-13:
also for modes you can mix 2 same channels

what i did was use gear switch towards me as angle so angle is always on.. and gear away is failsafe aka drop land or rth not sure if it rth or you need to set up rth as a seperate deal since that has that..

then aux 1 3 position switch is top gps position hold meaning holds still like a dji.. on the same gps switch still up i put headfree.. so same switch does 2 things hope that works might need to seperate that with another channel somehow..

mixing channels i havent done that in ages.. .. used to fly helis with none of this new tech on it.. boy has it changed

cdog4twenty, 2017-06-20:
Need a switch to reset home position, I was having the same issue until I set up a switch to reset home position. After enabling the reset I was able to clear the red x navigation

DzikuVx, 2017-07-04:
I assume problem is solved. Great. I'm closing it. If any new data appears on the topic, feel free to reopen if needed

closed 2017-07-04 by DzikuVx

Hennyr, 2018-03-23:
I have a matek F405-STD, M8N GPS with Compass, a DHOP of 2 and 12 sats, but no 'ALL GREEN'. Now am i looking for the 'Reset Home Position' I tried already the HP ans ALTHOLD before armin, but no succes.
Please inform me of possible errors

jakelynpot, 2018-05-02:
Can you help me with my issue? I cannot arm my quadcopter because the "Navigation is safe" indicator is lit red. I checked my CLI  by typing status bootlog tasks below. Please see result below. Also my GPS fix type is intermittently flashing  from 0 to the number of satellites available. How to fix this?

# status bootlog tasks
System Uptime: 324 seconds
Current Time: 2018-05-02T00:57:11.320+00:00
Voltage: 11.23V (3S battery - OK)
CPU Clock=168MHz, GYRO=MPU6000, ACC=MPU6000, BARO=BMP280, MAG=HMC5883
STM32 system clocks:
  SYSCLK = 168 MHz
  HCLK   = 168 MHz
  PCLK1  = 42 MHz
  PCLK2  = 84 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=OK, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=OK
SD card: None inserted
Stack size: 6144, Stack address: 0x10010000, Heap available: 1024
I2C Errors: 1, config size: 3404, max available config: 131072
ADC channel usage:
   BATTERY : configured = ADC 2, used = ADC 2
      RSSI : configured = ADC 3, used = ADC 3
   CURRENT : configured = ADC 1, used = ADC 1
  AIRSPEED : configured = none, used = none
System load: 1, cycle time: 1004, PID rate: 996, RX rate: 111, System rate: 9
Arming disabled flags: NAV CLI

#