iNavFlight/inav #1798: iNAV 1.6.1 Flayaway

domhan, 2017-06-13:
Hello,

I had a burning second last weekend. The Copter has flown without any problem - until now. At last weekand, I had a flyaway. My Setup:
- iNAV 1.6.1
- Omnibus F4 Pro
- M8N GPS

In the log I realised the switch of the "navState" variable from 5 to 7. I don't know the meaning. Maybe this is a possibility?

Attached you can find the Log-File. At about 1:09 the copter flyes away with full speed and full throttle. 

Thank you for your help. Best regards 
Gregor

[LOG00011.TXT](https://github.com/iNavFlight/inav/files/1070618/LOG00011.TXT)

miradarya, 2017-06-13:
You've got too much vibration on your copter, causing the accelerometer to read nonsense:
[image]

Try soft-mounting the flight controller and/or motors.

stronnag, 2017-06-13:
````
$ inav_modes.rb /tmp/LOG00011.TXT 
iNav version = 1.6.1 (states eq 1.6.1)
Time(s)	Elapsed(s)	State
  53.3	(   0.0)	nav_state_idle (1)
  64.1	(  10.8)	nav_state_poshold_2d_in_progress (5)
  69.4	(  16.1)	nav_state_poshold_3d_in_progress (7)
````
Commanded mode change?

digitalentity, 2017-06-14:
I see very common accelerometer resonance issue. What frame is this? Is the FC soft-mounted?

domhan, 2017-06-15:
OK, it seems to be some vibration. The only thing I changed are the Props. This was the first and last flight with 3-Blade-Props.  Maybe more virbation because of out-of-balance. I will test and report.

Thank you!

DzikuVx, 2017-09-05:
I think we can close it. In case of new data, please reopen or create a new issue

closed 2017-09-05 by DzikuVx