iNavFlight/inav #1800: CURRENT_METER_SCALE on OmnibusF4PRO (SD)

alfag4, 2017-06-15:
The default value is now 265 (in target.h) but measuring the current with an ammeter the correct scale for my omnibus seems 400. Why this new default value?

DzikuVx, 2017-06-15:
Probably different revision of the board. The ones we received from Airbot had 265 not 400. Which makes sense, since scale 400 gives max readout of about 82A at 3.3V

alfag4, 2017-06-15:
I bought it on airbot shop in march, so i don't know which version i have probably V3. In my case the value of 400 returns a current absorption identical to the ammeter.
Thanks.

DzikuVx, 2017-06-20:
OK, since looks like there are multiple types of this board on market (and clones and so on) and it is not possible to determine current meter scale from software, looks like we have to leave it as it is. Every user would have to calibrate it by himself

closed 2017-06-20 by DzikuVx