iNavFlight/inav #1803: LED strip control not working in inav_1.7.1_OMNIBUSF4V3-1

Key0nee, 2017-06-18:
Hello,
      I have been testing inav_1.7.1_OMNIBUSF4V3-1 obtained [Here](https://github.com/iNavFlight/inav/pull/1779)  LED strip control does not appear to be working.. I have tested the same board in the exact setup with Cleanflight v2.0.2 for OMNIBUSF4SD target and it works fine.

flounderscore, 2017-06-19:
In iNav the led strip needs to be connected to PWM5 (see
https://github.com/iNavFlight/inav/blob/master/docs/Board%20-%20Omnibus%20F4.md
).

Key0nee, 2017-06-19:
Copy that... 

Thanks..

closed 2017-06-19 by Key0nee

jflyper, 2017-06-20:
FYI, F4 V3 (and some variants) has PB6 connected to the designated LED strip signal pin.

Ref:
https://github.com/betaflight/betaflight/wiki/Board---OMNIBUSF4#v317-and-later-led-strip-pin-assignment-has-changed-for-omnibusf4sd-target

Key0nee, 2017-06-20:
@jflyper 
I have it working for the most part on PWM 5 output but I have noticed that assigning an led color for battery condition doesn't change regardless of battery level..
Also Is there a method for changing  the color assignments for modes