iNavFlight/inav #1804: Omnibus F4 v3 DFU Mode detection problems..  Not unique to iNav...

Key0nee, 2017-06-18:
Hello,
         Probably not a firmware issue but I thought posting here might save others time and frustration.  I am using an Omnibus F4v3 for my latest build and found that my laptop will not recognize the Omnibus in DFU mode with a GPS plugged in and powered up.  This also happens with the latest version of Cleanflight for OMNIBUSF4SD target.  Removing power from the GPS seems to allow proper DFU mode detection and driver configuration...

K

oleost, 2017-06-18:
GPS or the compass ? Sda/SLC lines?

Having compass connected without powering it up is often begging for
problems.

Key0nee, 2017-06-18:
@oleost 
  I guess I should have included this detail.. but for me its the GPS... I am using the NEO m8N's compass but, the problem seems to be the same with the Sda/Slc connected or not...

digitalentity, 2017-06-18:
It's the hardware limitation. Built-in bootloader activates on USB/UART1. If there is activity on UART1 bootloader doesn't enable USB DFU.

Key0nee, 2017-06-18:
Copy that....

closed 2017-06-18 by Key0nee