iNavFlight/inav #1806: Spektrum Serial Telemetry Receiver (SPM4649T) Interfacing

JulianGoesPro, 2017-06-18:
It looks like iNav hasn't jumped on the Spektrum Telemetry Hype Train :)
I am not sure how much work it is to use the code Betalight has for this bi-directinal protocol. 
[url]https://www.rcgroups.com/forums/showthread.php?2775731-Spektrum-Serial-Telemetry-Receiver-%28SPM4649T%29-Interfacing[/url]
Thanks!

martinbudden, 2017-06-19:
I'll look at doing this for the next release. @JulianGoesPro are you volunteering to do the testing? (I don't have any Spekturm hardware.)

JulianGoesPro, 2017-06-19:
Sure, I have an OmniBusF4ProV3 running iNav and an SPM4649T (setup as a normal SpekSat right now) and sure would help testing :)

JulianGoesPro, 2017-06-29:
On what UART do you want me to solder it to?

closed 2018-05-27 by stale[bot]