iNavFlight/inav #1809: SPRacing F3 OSD Not booting : 1.7.1

GTorreil, 2017-06-20:
Using a 'chinese' _SPRacing F3 "integrate OSD"_ with magnetometer removed : 
- **1.6** -> Everything is working
- **1.7.0** -> No tested yet (will do soon)
- **1.7.1** -> No way to connect to Configurator after flashing (Probably not booting). 

Is this problem known? I havn't found anything related to this...

tonyintn, 2017-06-21:
Not an issue with INAV I also use the SPRacing F3 with integrate OSD have three actually all of them work perfect.

GTorreil, 2017-06-22:
I have to try with these parameters. I usually use a lower baudrate for flashing, but I don't think that's the cause for that issue, because I still can flash 1.6 this way and it works ...

GTorreil, 2017-06-22:
@tonyintn do you have 1.7.1 running on your F3s ? And did you select the SPRACINGF3 target or something else ?

parnz, 2017-06-22:
I've SP Racing F3 Integrated OSD Deluxe version 
Flashed iNav 1.7.1 (target SPRACINGF3) Manual buadrate set to 25600 as stated in [manual](http://www.phasercomputers.com.au/wp-content/uploads/2016/08/sp-racing-f3-osd-manual.pdf)
Flashed [MWOSD 1.7.1](https://github.com/ShikOfTheRa/scarab-osd/releases) via MWOSD Gui

Everything work find.
Note: Do not connect anything to UART2 other wise OSD will not work.

GTorreil, 2017-06-22:
whoa, thanks, maybe it was as simple as changing the baudrate.... I definitly have to try it. I'll keep the thread updated.

tonyintn, 2017-06-22:
Sorry for the delay yes you need to select the SPracingF3 target.

GTorreil, 2017-06-23:
This worked! I don't know what happened there before... 
Sorry for that issue which turned out to be a user-support issue ... Thank you all anyway! 
**[CAN BE CLOSED]**

closed 2017-07-01 by DzikuVx