iNavFlight/inav #1810: ANYFC7 External Barometer

Gavsv8, 2017-06-20:
The ANYFCF7 from Banggood doesn't have an internal barometer so I connected an external BMP280 and HMC5983 to the I2C bus. When changing options in INAV with the released HEX  I could only get the magnetometer working.
The target needed to be modified to get the barometer  working. I modified it to the below which works in this particular configuration:
// External I2C BARO
#define BARO
//#define USE_BARO_MS5611
//#define USE_BARO_BMP085
#define USE_BARO_BMP280
#define BARO_I2C_INSTANCE I2C_DEVICE_EXT

oleost, 2017-06-20:
Should work directly if you buy the correct barometer and solder it where it is supposed to go.

Looks like your modification will break support for the baro the way it is supposed to get connected?

https://www.rcgroups.com/forums/showpost.php?p=37391391&postcount=13577

stronnag, 2017-06-20:
The other defines should remain; then the user can either add the surface mount if they have the skill / equipment, or external I2C.

Gavsv8, 2017-06-21:
If we could support both options that would be great. I can run some tests if required.

Gavsv8, 2017-06-21:
So I'v enabled all in the target.h as shown below:
// External I2C BARO
#define BARO
#define USE_BARO_MS5611
#define USE_BARO_BMP085
#define USE_BARO_BMP280
#define BARO_I2C_INSTANCE I2C_DEVICE_EXT

and also updated target.mk to:

TARGET_SRC = \
            drivers/accgyro/accgyro_spi_mpu6000.c \
            drivers/barometer/barometer_ms56xx.c \
            drivers/barometer/barometer_bmp280.c \
            drivers/barometer/barometer_bmp085.c \
            drivers/compass/compass_hmc5883l.c \
	    drivers/light_ws2811strip.c \
            drivers/light_ws2811strip_hal.c \
            drivers/pitotmeter_ms4525.c

and my external baro still works fine.

stronnag, 2017-06-21:
Indeed, that's how pretty much every other target has always worked.

All your PR needs to do is add BMP280 to the target.

sambas, 2017-06-22:
Does this break baro detection on boards that have onboard baro?

DzikuVx, 2017-06-22:
I'm affraid it breaks them. What creates a problem since code is unable to look for one device on different buses. I will be facing this issue on my next built since it will be AnyFC F7 with external BMP280 too...

Gavsv8, 2017-06-22:
okay, looks like I'll have to close and rely on custom hex for this, thanks everyone

DzikuVx, 2017-06-22:
No no no, please do not close it. It's another proof that sensor detection should be improved !

DzikuVx, 2017-06-22:
@digitalentity I'm affraid this requires us to make a new subtarget or rewrite sensor detection code... damn...

digitalentity, 2017-06-22:
@DzikuVx I have a hacky solution in mind. Will try to code it after the weekend after recovering from jet lag :smile:

DzikuVx, 2017-06-22:
Either way, since I want to connect baro to my F7, I'm just writing a new
target for it ;]

Gavsv8, 2017-06-23:
I've tested the ANYFCF7_EXTERNAL_BARO target and its works fine

digitalentity, 2017-06-23:
@DzikuVx please go ahead and merge #1815 as a temporary solution until we have something better coded

DzikuVx, 2017-06-23:
@digitalentity with pleasure.  I also created #1816 to remember about this one. And this ticket is now closed

closed 2017-06-23 by DzikuVx