iNavFlight/inav #1814: Improvement of Automatic Landing in RTH

raul-ortega, 2017-06-22:
Perhaps it is a crazy idea, but in my community we have been discussing about how to improve the precision in automatic landing trying to avoid crash with the objects arround the home position. Please, read this issue like that, some ideas that we hope to be no much dificult to implement.

The image bellow shows in some way what could be a better procedure for the automatic landing:

[image]

It is based on the actual procedure, but adding two new components: The circular descent is performed until the aircraft reaches a minimum altitude. After that, it navigates through waypoints  to line the runway (real or virtual) and lands.

I'll try to explain better the full procedure:

0.- Define the virtual runway:

- Previous to take off the user place the aircraft on a point that could be considered the runway header. He connect the battery and home position is set (waypoint H).
- The user switch on with the transmiter to a mode that could be called "RUNWAY mode". 
- Then, the user walks to another point with the plane in his hand, taking into account the direction of the wind, and also considering that it is a safe path for the landing procedure, and with a stick combination, or perhaps pointing the nose of the aircraft to the sky, sets a waypoint (WP0).

This could help iNav to know the length and direction of a "virtual" runway.

1.- Take off:

- The user arms the aircraft and launch it to the air following the "virtual" runway, and flies.

2.- RTH and LOITTER

- When the user, or failsafe system, switches to RTH mode, the aircraft flies towards to WP3, which coudl be calculated automatically taking into account where the home position is set, and a radius.

- The aircraft starts to loiter with the defined radius and descend. Descent speed is limited to nav_landing_speed when altitude is above nav_land_slowdown_maxalt.

3.- Approximation 

- When nav_land_slowdown_maxalt is reached, the aircraft flies to WP2 keeping the altitude constant.

- And continues flying to WP1 keeping the altitude constant, with the aim of line the runway.

- WP2 and WP1 are calculated automatically by iNav taking into account the position, direction, and size of the runway, as well as the altitude and speed needed/configured for the landing procedure.

4.- Landing 

- When WP1 is reached, vertical speed is scaled down to one fourth of nav_landing_speed at nav_land_slowdown_minalt.

- Once Home position is reached, motor is stopped, and the aircraft glides until it lands.

I hope I have expressed the idea well and could be implemented as an improvement or an alternative to the current procedure.

skaman82, 2017-06-22:
> The user arms the aircraft and launch it to the air following the "virtual" runway, and flies.

That could be problematic because the launching direction is often depending on the wind situation. If you have a location with predefined landing area, the chances that the "virtual runway" aligns with launch direction are very slim.

[image]

digitalentity, 2017-06-22:
Initially I think we can just come in for a landing in the opposite direction we launched. I.e. if we launched due North, we'll come in for a landing due South and will aim to land at the spot we launched from.

DzikuVx, 2017-06-22:
@digitalentity I'd rather say take off and land into the wind. So landing should be the same as take off

digitalentity, 2017-06-22:
@DzikuVx this must be configurable. Where I fly you usually don't have much choice - there's city behind and turning for a landing over the buildings is less safe than landing in the opposite direction you launched.

skaman82, 2017-06-22:
@DzikuVx yes it would make sense. But we know for sure that the launching direction is safe. So landing in opposite direction is safe also. In the picture above there is a road in the upper left corner. So if the plane would land in the launch direction, it would most likely crash into a moving car :)

DzikuVx, 2017-06-22:
So let me quote someone:

> moah settings

raul-ortega, 2017-06-22:
"Then, the user walks to another point with the plane in his hand, taking into account the direction of the wind, and also considering that it is a safe path for the landing procedure, and with a stick combination, or perhaps pointing the nose of the aircraft to the sky, sets a waypoint (WP0)."

I think it doesn't matter in which direction you launch the aircraft if you set this waipoint in the previous instant to take off, taking into account the direction of the wind (for example, let's supposed that I know that the wind will not change too much during next 20 minutes in my flying filed) ... With the home waypoint and this wp0 waypoint, the virtual runway is defined in the oposite direction of the wind. When landing are known the position, length and direction of the runway (which is the oposite direction of the wind).

digitalentity, 2017-06-22:
@raul-ortega yes, this is definitely a good procedure. However, what if user redefines "home" in flight using "HOME RESET" mode? What if user doesn't define WP0 and hit RTH? We need a solution that will work in all cases (or at least as many cases as possible).

raul-ortega, 2017-06-22:
@digitalentity if WP0 is not defined, the landing should proceeds by default in loitter, just like now.

raul-ortega, 2017-06-22:
Also, in case of HOME RESET, default procedure for landing wolud be executed.

hali9, 2018-06-21:
I think it is possible to approach without defining waypoints by defining a virtual point for the approach.
The landing route begins at the home point and is directed at homepoint.yaw.
These items are saved during arming or during HOME_RESET.
It allows to take off in any direction and land in any designated direction.
Scenario:
1. Turn on RTH.
2. When the height falls below maxalt during circulation, keep the maxalt height until you reach homepoint.yaw + 135deg (decision point)
3. Calculate the virtual approach point and lower the height around this point until you reach the minalt.
4. Keep the minalt height until you reach homepoint.yaw direction.
5. Land using the angle mode with roll = 0 and pith = dive_angle (as it is now).
[image]

wx4cb, 2018-12-04:
also something that just came up in a flying session today. if an aircraft has retracts, would it be possible to automatically extend them below a certain alt that way for some that are pushers, you don't get a prop strike (or worse, a smashed underslung camera gimbal.

bishko, 2022-10-31:
> @digitalentity I'd rather say take off and land into the wind. So landing should be the same as take off

Yes Pawel, but the wind can change in a stort time and definitely changes after a 1h flight for example.

closed 2023-10-27 by DzikuVx

bakwc, 2024-01-01:
Sorry, why does it closed? Is this feature implemented?

MrD-RC, 2024-01-01:
It will be in the future. It is currently being developed, before being merged in to the main branch.