iNavFlight/inav #1816: Allow sensor detection on multiple I2C buses

DzikuVx, 2017-06-23:
As seen in #1810 and also previously with REVO/AIRBOTF4 targets, we lack the possibility to looks for sensors on multiple I2C buses.
Sensor as a type is pinned to a bus and code fails to use even different sensor of the same type on different I2C bus.

We should either pin exact hardware to a bus or scan all buses in hope for success. 

In first solution instead of
`#define BARO_I2C_INSTANCE I2C_DEVICE_EXT`
 we would have
`#define MS5611_I2C_INSTANCE I2C_DEVICE_EXT`

In the second solution:
`#define BARO_I2C_INSTANCE I2C_DEVICE | I2C_DEVICE_EXT` (or something like that)

digitalentity, 2017-11-28:
Done for compass on REVO https://github.com/iNavFlight/inav/pull/2522

digitalentity, 2018-02-12:
Fixed with `busDevice` migration

closed 2018-02-12 by digitalentity