iNavFlight/inav #1817: How to turn on SONAR with iNav 1.7.1?

liutairan, 2017-06-23:
Just like another post, I cannot find SONAR configuration. I also tried to use CLI feature SONAR, but it shows that invalid command. Is it removed again?

liutairan, 2017-06-23:
I saw that sonar was back in 1.7, but cannot find it in the configurator.

stronnag, 2017-06-23:
have you specified the sonar hardware ?
````
set rangefinder_hardware = 
Allowed values: NONE, HCSR04, SRF10
````

johnsilvester, 2017-06-23:
@liutairan  Yeah! use @stronnag  command in the CLI to activate it!

stronnag, 2017-06-23:
Not sure if SURFACE mode is working though. A brave test volunteer?

johnsilvester, 2017-06-23:
I'll test it out in the next day or two

liutairan, 2017-06-23:
@stronnag @johnsilvester Thanks~ Just found it. I will test it when the weather get better.

closed 2017-06-23 by liutairan

nerezw, 2017-07-06:
I connected an HC SR04 to my spracingF3, entered the CLI command, SONAR tab turns red and nothing else.... any ideas?

DzikuVx, 2017-07-06:
@nerezw we are not wizards or prohets, you have to be much more specific. How you connected and what you entered. CLI `diff` maybe?

nerezw, 2017-07-06:
I have an spracingF3. I connected sonar trigger to IO2/5(yellow) and sonar echo to IO2/6(green) and 5volts Vcc on HC-SR04.
I entered :  # set rangefinder_hardware
and got
rangefinder_hardware = NONE
Allowed values: NONE, HCSR04, SRF10
and then

# set rangefinder_hardware = HCSR04
rangefinder_hardware set to HCSR04

and the save

the SONAR tab on the top of the screen turns RED
the SANAR tab in the sensors screen can not be selected

can any body help

DzikuVx, 2017-07-06:
`diff` please. HC-SR04 works with INAV 1.7.1 since I tested it few days ago. So looks like you are doing something wrong after all

nerezw, 2017-07-06:
Ok... Will have a close look

nerezw, 2017-07-06:
here is the diff when i go into cli
Am I missing anything

Entering CLI Mode, type 'exit' to return, or 'help'

# diff

# version
# INAV/SPRACINGF3 1.7.1 Jun  3 2017 / 08:13:23 (48e1feff)

# resources

# mixer

# servo

# servo mix

# feature
feature -RX_PPM
feature RX_SERIAL
feature GPS

# beeper

# map

# serial
serial 1 2 115200 9600 0 115200
serial 2 64 115200 38400 0 115200

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# master
set acc_hardware = MPU6050
set acczero_x = 34
set acczero_y = 3
set acczero_z = -188
set accgain_x = 4098
set accgain_y = 4090
set accgain_z = 4082
set rangefinder_hardware = HCSR04
set mag_hardware = HMC5883
set mag_declination = -820
set magzero_x = -73
set magzero_y = -68
set magzero_z = 82
set baro_hardware = MS5611
set serialrx_provider = IBUS

# profile
profile 1

#

johnsilvester, 2017-07-06:
Try soldering to the 'Trig' and 'Echo' pins on the bottom side of an spf3

nerezw, 2017-07-06:
Will do... Thanks

liutairan, 2017-07-07:
@nerezw Have you solved your problem? Did you use usb port to connect to the computer? If so, maybe you can try providing extra power to the board. At least for me, the board cannot provide enough power for the sonar if using USB connecting to computer. Once I have the battery connected or other power source connected, the sonar will work well; once I remove the battery, the sonar indicator will get red again.

nerezw, 2017-07-08:
I haven't solved it yet.  I was going to try 1k resistors on tabs under the
board as suggested by someone

matevz25, 2017-09-01:
On cleanflight sonar works ok with 1k resistor on sda and sdl. I tried with inav cli enabled, doesnt work its red.