iNavFlight/inav #1820: Navigation isn't safe

millioner2, 2017-06-24:
Hello,
I hope someone can help me.

Please take a look at the pictures and tell me why the navigation is not safe.

https://feuerwehr-nuembrecht.de/images/1.jpg
https://feuerwehr-nuembrecht.de/images/2.jpg
https://feuerwehr-nuembrecht.de/images/3.jpg

thank you in advance

parnz, 2017-06-24:
see #1797

oleost, 2017-06-24:
Safety will block arming in any of the following conditions:

You configured WP/PH/RTH or Failsafe RTH and you don't have a good accuracy GPS fix
You try to arm into RTH/PH/WP
You have waypoint mission in memory and your first waypoint is too far from your current position
Given it's a small NZ mini GPS my guess is that accuracy is not good. You need at least 6 sats and position accuracy better than 9 meters (usually HDOP of 5-6 or lower).

millioner2, 2017-06-24:
@oleost 
You are right I use an NZ mini GPS. What kind of GPS would you recommend?

oleost, 2017-06-24:
Show a screenshot of gps tab

millioner2, 2017-06-24:
@oleost 
https://feuerwehr-nuembrecht.de/images/4.jpg

stronnag, 2017-06-24:
Note the number of 'Sats'.

Timeout count does not bode well either.

millioner2, 2017-06-24:
Ok the modul is not good I understood but what would they recommend?

oleost, 2017-06-24:
You need more than 6 satellites and proper reception. 

1. Go outside
2. Shield gps from electronics noise ( put it on a mast longer away from rest of electronics )
3. Buy new GPS if all other fails.

Closing this support issue. If you have more questions please use rcgroups forums

closed 2017-06-24 by oleost

parnz, 2017-06-24:
for recommended GPS read here [GPS and Compass setup](https://github.com/iNavFlight/inav/wiki/GPS--and-Compass-setup)

asteman, 2017-10-13:
@oleost I'm on 1.7.3. You mentioned that arming isn't possible with WP set "too far off". I can set a waypoint 50m off. But not a few 100m or 10km. What is the definition of "too far off"?

stronnag, 2017-10-13:
The value set in the CLI by the key `nav_wp_safe_distance`. The default is 10000( cm) i.e. 100m.