iNavFlight/inav #1821: Post-1.7.1 development blackbox tidy creates erroneous log files.

stronnag, 2017-06-24:
Log files post the bb tidy changes https://github.com/iNavFlight/inav/pull/1751 https://github.com/iNavFlight/inav/pull/1752 have erroneous data, including (but possibly not limited to) the following when extracted with `blackbox_decode`

* Erroneous sensor failure warnings
* Invalid GPS positions (these look like deltas from a last known point)

Replays (LTM v BB) and an example log file have been provided in https://github.com/iNavFlight/inav/pull/1752#issuecomment-310763522

stronnag, 2017-06-24:
Here's an example from another vehicle (the normally mega reliable tricopter), HW sensor "failures" not reported via LTM. This machine was previously flown on 78e6431a (14 June) with no such error reports (position or hw status). So the first set of BB tidy does not seem to cause this issue.

Also, and in common with the log in #1752 (comment), intermittent strange GPS coordinates, e.g.
````
2017-06-24T09:07:30+0100 Ignore bogus 0.000076 0.000226 (50.910663 -1.534713)
2017-06-24T09:07:30+0100 Ignore bogus 0.000053 0.000206 (50.910663 -1.534713)
2017-06-24T09:07:31+0100 Ignore bogus 0.000029 0.000189 (50.910663 -1.534713)
````
time stamp bogus lat lon (last good lat lon)

HW status changes:
````
$ ./inav_hw_status.rb //t/inav/2017-06-21_tri/LOG00005.TXT 
iNav version = 1.7.1 (states eq 1.7.1)

0.000s HW Status change (155 341)
gyro	OK
acc	OK
mag	OK
baro	OK
gps	OK
rangef	none
pitot	none

12.046s HW Status change (d 13)
gyro	OK
acc	unhealthy
mag	none
baro	none
gps	none
rangef	none
pitot	none

14.506s HW Status change (155 341)
gyro	OK
acc	OK
mag	OK
baro	OK
gps	OK
rangef	none
pitot	none

20.594s HW Status change (33 51)
gyro	unhealthy
acc	none
mag	unhealthy
baro	none
gps	none
rangef	none
pitot	none

22.051s HW Status change (155 341)
gyro	OK
acc	OK
mag	OK
baro	OK
gps	OK
rangef	none
pitot	none

28.205s HW Status change (11 17)
gyro	OK
acc	none
mag	OK
baro	none
gps	none
rangef	none
pitot	none

29.209s HW Status change (155 341)
[... 830 lines cut ...]
mag	OK
baro	OK
gps	OK
rangef	none
pitot	none

594.093s HW Status change (53 83)
gyro	unhealthy
acc	none
mag	OK
baro	OK
gps	none
rangef	none
pitot	none

594.546s HW Status change (155 341)
gyro	OK
acc	OK
mag	OK
baro	OK
gps	OK
rangef	none
pitot	none

````

martinbudden, 2017-06-25:
@stronnag , PR https://github.com/iNavFlight/inav/pull/1825 should fix this. Can you test?