iNavFlight/inav #1822: INAV m8n gps toilet-bowling issue

94onur94, 2017-06-24:
Hi guys, my pos hold issue continues. When I switch on nav poshold, it rounds around holding position for one time and holds in the air at the right position. 

My SP Racing F3 and ublox m8n gps use HMC5883L compass. I tried both compasses separately but problem is same.

About compass calibration: I'm in the Northern Hemisphere Ankara Turkey, Z-component is positive - when pitching the vehicle down, the Y component increases in value - when rolling the vehicle right, the X component increases in value. I calibrated it in different times and values are same. Also I set magnetic declination value for Ankara, Turkey(+5° 37'). I think the problem is not about calibration.

My issue video: https://youtu.be/AGe2rb-9Ufo

F450 Frame
Sp racing f3 deluxe controller
Dji 2212/920kv motor
9450 propeller
Ublox Neo-M8N GPS
Flysky fs-i6 transmitter / fs-ia6b receiver
3s 5200mah lipo

Dump: https://pastebin.com/9GSWajvr

oleost, 2017-06-24:
You will need to physically disable your on board compass. Either rip it from board or cut the correct.

Search Google, rcgroups forum etc and you will find information on how to do it.

94onur94, 2017-06-24:
@oleost But why does it continue when I disconnect m8n compass. I tried only sp racing f3 compass, it did same thing.

oleost, 2017-06-24:
https://github.com/iNavFlight/inav/wiki/GPS--and-Compass-setup

DzikuVx, 2017-06-24:
@94onur94 you can not have 2 compasses connected at the same time. It does not work that way. To use external one you have to get rid of that onboard of SPRF3. By get rid I mean remove it from the board.

Ralf-W, 2017-06-24:
I am not oleost but here is my answer to "But why does it continue when I disconnect m8n compass. I tried only sp racing f3 compass, it did same thing."
The onboard compass is too close to magnetic fields developed by the copter itself: Cables to the motors, buzzer, screws, etc. This makes the MAG send wrong data that produce toilet bowling. That's why onboard compass should not be used in a copter. 
As your external MAG is of the same type you have need to physically disable your on board compass - as ole wrote above.

94onur94, 2017-06-24:
Users says same things in all answers so I will disable on board compass on SP Racing F3. I hope it will fix. Actually it has to differentiate on board compass from external compass. Maybe in future upgrades it might come.

stronnag, 2017-06-24:
It is impossible, both devices share the same address. Perhaps in future you will take the advice of other users and not waste developers time with support questions.

DzikuVx, 2017-07-01:
Since there seems to be no development here and this is obviously a support issue, I'm closing it. If any new data appears, feel free to reopen

closed 2017-07-01 by DzikuVx