iNavFlight/inav #1823: SmartAudio not Supported!!!!!!!!!!!!!!!

Azobe, 2017-06-24:
iNav is such a groundbreaking amazing controlling software. When I first heard of it I felt like it expanded the universe of all the SMT32 Flight Controller Boards to whole new horizons and got me super excited. 

I bought an Airbot Omnibusf4v3, GPS, the whole deal, with the vision to build this perfect small 6" Chase/Acro/Long Distance quadcopter. I chose TBS Unify Pro HV 5G8 as it had amazing flexibility, 800mW of transmission power, and in-OSD configuration options - all of which would help realise my dream angel quadcopter. 

But Lo and behold, iNav doesn't support Smart Audio!!!! :( 
Like an didn't realise this until I was halfway through my build. And to be honest I am super surprised as well, as iNav carries over so much of BetaFlights capabilities - like Betaflight on steroids. 

Anywho, my dream quadcopter project is on hold for now because of this. This is my plead to all the talented iNav/Betaflight developers out there to port over SmartAudio functionality into iNav. I am a sub-mediocre programmer at best and am not familiar with the ins and outs of the sophisticated programming that goes behind these flight controllers. But I would be immensely grateful if this feature would come aboard iNav in a reliable fashion soon, and legitimately realise my FPV dreams.

oleost, 2017-06-24:
INAV is fantastic, your right.

Regards to smartaudio it's already being worked on.

Read this. And I'm closing this issue.

#1262 #1576

closed 2017-06-24 by oleost

Azobe, 2017-06-24:
roger