iNavFlight/inav #1824: FW: blackbox rcCommand rate do not match axis rate

DzikuVx, 2017-06-24:
@digitalentity I observed something strange in 1.7.1 on fixed wing: rccommand rate do not match axis rate. Look here:

[image]
[image]

Roll rate set to 300dps, rcCommand requests 430, Iterm behaves like it wants to negate missing 130dps. Either log is wrong or rcCommand is not computed correctly and/or Iterm acts in wrong direction. I just do not get it...

[wing_v3_blackbox_log_2017-06-24_205106.TXT](https://github.com/iNavFlight/inav/files/1099969/wing_v3_blackbox_log_2017-06-24_205106.TXT)

digitalentity, 2017-06-25:
Blackbox explorer maps rcCommand to [deg/s] incorrectly. It's more correct to observe `axisRate` variable which is PID controller setpoint. In your case at that specific point in time gyro rate is 348 deg/s and `axisRate` is merely 292 deg/s, so the wing overshoots the desired rate - that's why you see negative `P` and decreasing `I` - PI-part of the controller is compensating for overshoot.

DzikuVx, 2017-06-25:
ok, thanks, moving to BB explorer project

DzikuVx, 2017-06-25:
iNavFlight/blackbox-log-viewer#6

closed 2017-06-25 by DzikuVx