iNavFlight/inav #1829: Horizon drift when OSD is used

digitalentity, 2017-06-25:
Scenario:

1. Fast cycle time (1 kHz and above) on a relatively slow CPU (F3, OMNIBUS)
2. Enabled OSD with a lot of data updating (GPS coordinates, AHI, speed, altitude)
3. Rapidly changing attitude (i.e. airplane launch)

Result: Attitude drift

Cause: OSD screen update takes too long and IMU update time is jittery - this is known to cause incorrect attitude estimation.

Possible ways to reduce the effect:

1. Use slow looptimes (500Hz)
2. Use smaller `MAX_CHARS2UPDATE` (side effect - possible issues with "ghost" OSD characters)
3. Fix OSD code to use DMA (not working currently)

Relevant issues: #1776 #1445

marbalon, 2017-06-25:
DMA can be used only for targets with OSD chip on separate SPI bus. But MAX_CHARS2UPDATE can be reduced even to 20 and should not be noticeable (I used this value in the past without any issue).

OSD driver task is set to 60Hz, and inside layout is refreshed 1/5 cycles so 12Hz and driver takes the reset 48Hz. 20 chars every driver idle = 960 chars per second. And we should keep in mind that driver not all screen but only changed chars on the screen.

martinbudden, 2017-06-25:
It would be interesting to get some timings to see if it is the SPI write to the MAX7456 chip or the drawing of the OSD elements (ie the refreshing of the display layout) that is causing the jitter.

My suspicion is that it is the drawing of the OSD elements.

As @marbalon states, the way it currently works is that `osdUpdate` is called at 60Hz. In 4/5 calls to `osdUpdate` 20 characters are written to the MAX7456 chip, and in the other 1/5 calls *ALL* of the OSD elements are refreshed.

I think a better method would be to call `osdUpdate` more frequently, say 100Hz (or possibly more), and in every call update 1 OSD element and (say) write 10 characters to the MAX7456 chip.

If there are 20 OSD element on screen, then every element will get updated at 5Hz. We could get more sophisticated and have a priority scheme so that slowly changing elements (eg craft name, arm state etc) get refreshed less often, but that's probably not worth the effort.

marbalon, 2017-06-25:
The problem is that split refreshing elements is hard to do because at start all chars in shadow buffer is cleared and then all the elements are rendered because shadow buffer need to be ready before next driver update. If we want to draw element one by one every element need to clear space on the screen used by it self and then draw new content.

I've just measured how much time take osdRefresh() - about 130us for my default setup in planes (17 elements), and max 260us if I enable all elements including PID's etc.

max7456DrawScreen() can take up to 200us depends on how many chars need to be updated (MAX_CHARS2UPDATE  = 100)

@digitalentity detect that this is related with max7456DrawScreen() when more chars are written by driver so first we can test solution with smaller MAX_CHARS2UPDATE.

martinbudden, 2017-06-25:
@marbalon 
> If we want to draw element one by one every element need to clear space on the screen used by it self and then draw new content

Yes, I mentioned that. This is not that difficult. Many of the screen elements are fixed size anyway, so naturally self-clear. For the others it's just a question of knowing their max size and clearing it (potentially some of them could be made fixed size as well). Then there are one or two that will require a bit more work, eg the artificial horizon. None of this is particularly difficult.

Your timings - for which FC are they? F3 or F4?

marbalon, 2017-06-25:
Timings comes from my board KFC32F3 - so it is F3 ;-)

digitalentity, 2017-08-28:
@marbalon @raul-ortega is the issue still relevant in latest development. I don't see any issues with horizon since #1831 was merged.

digitalentity, 2017-09-04:
Closed due to no further reports of the issue. I also can't reproduce the issue anymore.

closed 2017-09-04 by digitalentity