iNavFlight/inav #1832: ANYFC F7 SD CARD NOT DETECTED

flyfisher2, 2017-06-26:
There is a comment in the target.h file that Bangood used the wrong sd card socket and line 116 seems to have been added to compensate for this:        #define SDCARD_DETECT_INVERTED

In my board, the hardware seems to have been corrected and it is therefore necessary to remove line 116 to get the card  detected.

(On the board I have the card present switch is normally closed ie it opens when a card is inserted.)

DzikuVx, 2017-06-28:
@flyfisher2 which AnyFC F7 is this exactly? "Pro"? Post a link please

flyfisher2, 2017-06-28:
Hi,
No, the one I have is this one:

https://www.banggood.com/STM32F745-100lqfp-216MHz-MPU6000-SPI-F7-Flight-Controller-for-FPV-Racing-Support-Betaflight-p-1137386.html?rmmds=detail-top-buytogether

DzikuVx, 2017-06-29:
OK, just checked on mine (the same) and it does not works either. Looks like we will need a special target for those clones. I will take care of that

DzikuVx, 2017-06-29:
@flyfisher2 please check if #1847 solves the problem. CLI setting `sdcard_detect_inverted` now allows to set SD card detection method via CLI

flyfisher2, 2017-06-29:
Hi,
OK, will check this evening when I get home.
Cheers

flyfisher2, 2017-06-30:
Hi,
No problems. that works really well.
BTW for people who (like me) enjoy the comfort of a buzzer, I have changed line 27 in the target.h file to output a trigger for an external buzzer with a pnp driver stage on output 4.
#define BEEPER   PB5

DzikuVx, 2017-06-30:
Great. About buzzer, take a look here: #1846 Buzzer will be put on Output nr.9 since this is least used output.
And since implementation are in place, I'm closing this one

closed 2017-06-30 by DzikuVx

flyfisher2, 2017-06-30:
Do you know if iNav will run on the PRO version as it only has 8 outputs and motor 1 is on output 10? This could also be problem with the buzzer on output 9.

DzikuVx, 2017-06-30:
@flyfisher2 I've ordered PRO version to check. I have no idea how this is wired, but I suspect that in PRO version they removed outputs 8 and 9 and changed order of the rest. But without the board its hard to know.
But, that Pro version has a separate buzzer output and I suspect this is wired to PB2 which is not connected anywhere on non-Pro versions. When I will know for sure, I will make that work :)

DzikuVx, 2017-06-30:
@flyfisher2 final question. You had to use `sdcard_detect_inverted` ON or OFF to make that work?

flyfisher2, 2017-06-30:
Hi,

I can see a B+ and B- on the bg photo, but I think that this could be Vbat and not  a buzzer output.

I think that you are probably correct about reordering the outputs.  Betaflight also uses output 10 for motor 1, so they may have tracked motor 1 to output 8.

I am also going to order the PRO version and would be very grateful for any input from you ( I am a hardware type, any software assistance in words of one syllable is gratefully received).

My direct email is    iw1904@bigpond.net.au <mailto:iw1904@bigpond.net.au> 

 

 

from: Paweł Spychalski [mailto:notifications@github.com] 
Sent: Friday, 30 June 2017 7:28 PM
To: iNavFlight/inav <inav@noreply.github.com>
Cc: flyfisher2 <iw1904@bigpond.net.au>; Mention <mention@noreply.github.com>
Subject: Re: [iNavFlight/inav] ANYFC F7 SD CARD NOT DETECTED (#1832)

 

@flyfisher2 <https://github.com/flyfisher2>  I've ordered PRO version to check. I have no idea how this is wired, but I suspect that in PRO version they removed outputs 8 and 9 and changed order of the rest. But without the board its hard to know.
But, that Pro version has a separate buzzer output and I suspect this is wired to PB2 which is not connected anywhere on non-Pro versions. When I will know for sure, I will make that work :)

flyfisher2, 2017-06-30:
As expected, sdcard_detect_inverted = OFF