iNavFlight/inav #1833: CRSF Telemetry not picking up RSSI/Link quality.

peeejayz, 2017-06-26:
As title, when running 2 way CRSF telemetery/control link inav doesn't pick up the RSSI/Link quality to put this into the OSD/UI

Dronek, 2017-06-28:
RSSI, Link Qly and SNR are handled by the crossfire TX module, AFAIK.
In my case, all parameters are recognized by the Transmitter plus additional telemetry coming from inav.

You can send the RSSI (or combined RSSI/LQ) value as PWM over a free channel to iNav and set iNav to use this channel for incomming RSSI information. In order to do so, you have to map the respective channel via the crossfire module menu.

In my case, RSSI on PWM channel is correctly recognized by iNav and shown in the configurator as RSSI value **but my minimOSD is always stuck at RSSI = 100% despite being configured to use FC RSSI value.**
I don't know if this is related to iNav or mwosd but it was certainly working for quite a while.

digitalentity, 2017-09-08:
Can't reproduce. RSSI/LQ is injected into RC channel correctly and picked up by INAV

closed 2017-09-08 by digitalentity