iNavFlight/inav #1834: Sp Racing F3 Evo ANGLE: motor spin when armed

monelli, 2017-06-26:
Hello to all. 
I try Inav but i have a problem. 
I use SP Racing F3 EVO on 500 quad with 12 inch prop.
I load a 12inch preset. I dont use GPS for now.

First test with prop, in ANGLE mode, i ARM, with minimum throttle and quad take off. For landing i DISARM. 

Attached some logs and a video without propellers. 
Why motor minimun is 1150? Visualization BUG?
Why in ANGLE two engines running? 
Is there a bug or is there a configuration problem?

For the first flights I used betaflight. Excluding the pid configuration and the quad too nervous I could fly normally...
[Diff.txt](https://github.com/iNavFlight/inav/files/1103239/Diff.txt)
[Dump.txt](https://github.com/iNavFlight/inav/files/1103240/Dump.txt)
[LOG00008.TXT](https://github.com/iNavFlight/inav/files/1103241/LOG00008.TXT)
[Stat.txt](https://github.com/iNavFlight/inav/files/1103242/Stat.txt)

https://www.youtube.com/watch?v=o7yvOvjVY4w

giacomo892, 2017-06-27:
Calibrate ESCs and tune PIDS, disable internal magnetometer and it should fly ok.

monelli, 2017-06-27:
ESCs are calibrated.
Mag in ANGLE is not used. Correct?

How should I correct PIDs?

I repeat, now in ANGLE, with minimal throttle, the quad goes vertically, it is impossible to land. (throttle stich is at miminum).

stronnag, 2017-06-27:
you might try with
````
set async_mode = NONE
````

DzikuVx, 2017-06-27:
I strongly suspect the reason are excessive vibrations. And this is quite strong suspection. @monelli  please record a blackbox log and share it so we can confirm

DzikuVx, 2017-06-27:
Ah no, there is a log, let we take a look

monelli, 2017-06-27:
Log is without prop! 
Fc is soft mounted.
With betaflight 3.2.0 with stock pid i flight, no problem.

DzikuVx, 2017-06-27:
Then please check wiring if throttle is indeed on radio channel 3. What Configurator Receiver tab gives? All sticks are correctly mapped? This is PPM with how many channels? Are you able to switch to PPM

monelli, 2017-06-27:
All stick are ok. I use only one cable, Serial RX, SBUS.
Receiver Tab is ok.
See video throttle stick is ok.

GTorreil, 2017-06-28:
Is the problem that the quad does not descend at all when stick at min ? if it is, try different PID values, perhaps a higher P ; this happened to me some times when flying with mis-tuned PIDs.

monelli, 2017-06-28:
Inav and Betaflight have comparable PID values?
I've seen some video and read documentation but I do not know how to set PIDs.

I explain:
With betaflight I used the following stock values and I could fly. The quad was too reactive.
> set p_pitch = 58
> set i_pitch = 50
> set d_pitch = 35
> set p_roll = 40
> set i_roll = 40
> set d_roll = 30
> set p_yaw = 70
> set i_yaw = 45
> set d_yaw = 20
> set p_level = 50
> set i_level = 50
> set d_level = 75

I then tested INAV with the PID presets for 12inch
> set mc_p_pitch = 100
> set mc_i_pitch = 30
> set mc_d_pitch = 25
> set mc_p_roll = 100
> set mc_i_roll = 30
> set mc_d_roll = 25
> set mc_p_yaw = 120
> set mc_i_yaw = 45
> set mc_d_yaw = 0
> set mc_p_level = 10
> set mc_i_level = 7
> set mc_d_level = 75

For me P value It's already high ...

digitalentity, 2017-09-08:
No activity. Closing.

closed 2017-09-08 by digitalentity

eziosoma, 2018-09-09:
Hi monelli, how do you resolved the problem?