iNavFlight/inav #1839: Refactor smix rules

DzikuVx, 2017-06-28:
Current way how `smix` works does not makes much sense. Some elements are not needed, other are duplicated. So:

* [ ] rename `rate` to `weight`
* [ ] remove `min` and `max` from `smix` so it works as `smix <target> <source> <weight> <speed>`
* [ ]  consider removal of `smix reverse` since the same can be done by applying negative `weight`

Overall it should give few bytes of Naze too...

DzikuVx, 2017-06-28:
@digitalentity it turned out that we can ntot get rid of `smix reverse` since non-custom mixers are not really loading custom `smix` rules.

And that gives us a very interesting question: why do we have custom and non-custom mixers, he? Why not to try to allow to load `smix` and `mmix` always? If custom mixes are predefined, then they would just replace predefined and whole thing would just work like custom. The way I see it, it's interesting idea

digitalentity, 2017-06-29:
Ok, understood. 
Reason for having custom and non-custom mixes is to have different mixer configurations integrated into the firmware. Non-custom mix doesn't load user-definable rules - this is reasonbale, otherwise we would be essentially leaving only custom mixes.

DzikuVx, 2017-06-29:
And here we come to the essence of my question about custom mixers. Here is how I see it (will be longish)

1. The only difference between Airplane and Airplane Custom is that Airplane Custom tries to load custom mmix and smix
1. If you look how loading of servo smix looks, you will notice it tries to empty current array and replace with smix
1. Now, _what would happen_ if decision if to load mmix and smix was dependent on the fact **if those rules exists**. Then, CUSTOM mixer would become obsolete. I set mixer type = airplane. As long as I do not define mmix and smix, I have regular airplane. When I enter mmix or smix it starts to behave like CUSTOM mixer type. There would be no differentce between them at all. The same goes for custom quad and tri
1. If to looks at it, it might have had sense in MultiWii times. Now it does not. At all. If we allow all mixer types to load mmix and smix (and replace predefined ones) we have a nice simplification here

But this is not all. I've been thinking about it, and the way I see it, we do not need mixer table like we have right now at all. We inherited it from MultiWii and over there it maybe had some sense.

Bottom line is that we have to provide following information:

* Type of device: multirotor, airplane, tricopter (?), rover (rover and boat)
* mixer rules for motors and servos

And here I ask: why firmware should "know" from the beginning how Octa Flat should work? We select Octa Flat from configurator after all. For the firmware, knowledge how all predefined mixers looks like is really not needed. It only has to know about `current` configuration.

So, on the longer run, I would remove predefined mixers, replace them with `device type` and store all predefined mixers in Configurator. After changing "mixer" in configurator, mmix and smix rules would be applied together with device type information.

And we would save some flash memory as well by the way...

digitalentity, 2017-06-29:
I believe we already had this discussion when we decided to disable "uncommon" mixers and concluded that having it defined as "platform type" in the firmware and "mixer" in configurator is good, but decided not to implement it at that time.

I think we should leave it as is now (including custom mixer types) - otherwise it may cause havoc if somebody has custom mixer from i.e. quad, but mixer type for airplane (happens when I move the board around from one craft to another).

martinbudden, 2017-06-29:
I agree that the predefined mixers should eventually be moved to the configurator. Not this release though.

digitalentity, 2017-09-08:
@DzikuVx any news here?

DzikuVx, 2017-09-09:
Nope, let's leasve it open. Probably during winter I will have enough time to finally move mixer rules to Configurator

DzikuVx, 2018-05-09:
OK, I'm closing this one since majority of work is already done. Decission if to remove `smix reverse` will be done separately

closed 2018-05-09 by DzikuVx