iNavFlight/inav #1844: unable to get gps modes to work

thewolfbird, 2017-06-29:
Hi 
I'm trying to get gps modes working, but it wont enable in modes tab. RTL should enable angle, althold automatically.
I have gps with at least 17 satellites on softserial 2 and it's working fine, can see gps modes appear in modes tab. All green lights ticked in setup tab, acc calibrated, etc
I can arm the plane no problems,   angle mode highlighted and working well, but when I  try and switch into RTH or poshold nothing happens. 
I have the exact same setup on another plane, the switches are all using the correct aux channels.
Maybe it's some parameter that I have missed, 
CLI

# dump#
# INAV/NAZE rev 6 1.7.1 Jun  3 2017 / 08:06:59 (48e1feff)
mixer AIRPLANE

mmix reset

servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1500 100 -1 
servo 3 1000 2000 1500 100 -1 
servo 4 1000 2000 1500 -100 -1 
servo 5 1000 2000 1500 100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 
smix reset

feature -RX_PPM
feature -VBAT
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature RX_PPM
feature VBAT
feature MOTOR_STOP
feature SOFTSERIAL
feature GPS
feature PWM_OUTPUT_ENABLE
beeper GYRO_CALIBRATED
beeper HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
beeper GPS_STATUS
beeper RX_SET
beeper ACTION_SUCCESS
beeper ACTION_FAIL
beeper READY_BEEP
[... 269 lines cut ...]
set nav_mc_pos_z_d = 0
set nav_mc_vel_z_p = 100
set nav_mc_vel_z_i = 50
set nav_mc_vel_z_d = 10
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set tpa_rate = 0
set tpa_breakpoint = 1500

digitalentity, 2017-06-29:
You are trying to switch to poshold/RTH in the air, right?

thewolfbird, 2017-06-29:
no from my table, as I want to test the modes

digitalentity, 2017-06-29:
As listed in the wiki here: https://github.com/iNavFlight/inav/wiki/Navigation-modes :

thewolfbird, 2017-06-29:
so everything is perfectly fine,  poshold and RTL should enable once in the air. I've got the RTL to work from the bench on my other plane and quad,  can see it highlight the other modes when enabled.

closed 2017-06-29 by thewolfbird

locutus40, 2017-09-24:
mine dont work in the air at all.. flies fine but once i hit my switch i set up for alt hold and pos hold it drops alittle and does nothing same thing as this guys.. any ideas?c

my dump
received - 0x2200344334570a20383633
2017-09-24 @ 04:34:05 -- CLI mode detected
Setup
Presets
Ports
Configuration
Failsafe
PID tuning
Advanced tuning
Receiver
Modes
Adjustments
Servos
GPS
Motors
OSD
LED Strip
Sensors
Tethered Logging
Blackbox
CLI
Note: Leaving CLI tab or pressing Disconnect will automatically send "exit" to the board. With the latest firmware this will make the controller restart and unsaved changes will be lost.

Entering CLI Mode, type 'exit' to return, or 'help'

# dump

# version
# INAV/OMNIBUS 1.7.3 Sep  9 2017 / 10:09:29 (974440df)

# resources

# mixer
mixer QUADX

mmix reset

# servo
servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1500 100 -1 
servo 3 1000 2000 1500 100 -1 
servo 4 1000 2000 1500 100 -1 
servo 5 1000 2000 1500 100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 

# servo mix
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
[... 486 lines cut ...]
set nav_mc_vel_z_i = 50
set nav_mc_vel_z_d = 10
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 50
set pitch_rate = 50
set yaw_rate = 45
set tpa_rate = 0
set tpa_breakpoint = 1500

#

digitalentity, 2017-09-26:
@locutus40 please clarify what is "does nothing"? Try setting "set nav_use_midthr_for_althold = ON" and moving throttle stick to the middle after you activate AltHold.