iNavFlight/inav #1848: Question -will enabling magnetometer for fixed wing create any issues?

thewolfbird, 2017-06-29:
Currently flying without mag for fixed wing, works well, but if I enable mag will this create any issues. 
Anyone flying with mag enabled for fixed wing? is it safe to use?

thewolfbird, 2017-06-30:
I've purchased the external gps with mag as recommended by inav, does anyone know the correct set align_mag  value
 set align_mag = CW270FLIP seems to work, but a bit slow

Redshifft, 2017-07-02:
Anytime you can avoid using Mags do so and don't look back. they are without doubt the most troublesome component, why do you want to use it ?

thewolfbird, 2017-07-02:
Just wanted to see if it works, but from your experiences using the mag, I may not want to now. I have it setup with my quad and it works ok.

closed 2017-07-28 by DzikuVx