iNavFlight/inav #1850: OMNIBUS F4 V2 GPS fix takes too long

assael07, 2017-07-01:
Hi, I just started using INAV few months ago and I found the GPS fix takes more than 3minutes after powering ON the quadcopter. 
I used the same GPS on APM with same GPS placement and it could get GPS fix just within 40seconds.
Any idea about this?

krzysztofmatula, 2017-07-01:
It looks like your new controller is disturbing your GPS.
STM32 F1/F3 (not sure about F4/F7) based controllers are known to interfere with GPS modules.
The only solution is to mount the GPS module in a further distance from the controller and do some shielding. Try googling.

assael07, 2017-07-02:
Hi krzysztofmatula,
Thanks for your quick response. 
Yes, it seems there is noise from the omnibus f4 which interfere GPS signal.
After adding grounded aluminium shield to the omnibus f4, I can get GPS 3D fix just within 30-40 seconds.

Redshifft, 2017-07-02:
Frsky receivers along with many other things will interfere with GPS reception so always keep components separated and screened as much as possible.

DzikuVx, 2017-07-14:
Since there is no new data and it seems problem has been solved, I'm closing this. Thanks

closed 2017-07-14 by DzikuVx