iNavFlight/inav #1851: PPM input not working on SPRACINGF3

krzysztofmatula, 2017-07-01:
Just checked latest development (76f077f) and the PPM input is not working at all on SPRACINGF3.
Also buzzer keeps beeping and warning leds blink green.

Going back to 1.7.1 all is fine.

My diff:
```
# diff

# version
# INAV/SPRACINGF3 1.7.2 Jul  1 2017 / 16:44:01 (76f077fa)

# resources

# mixer

# servo

# servo mix

# feature
feature SOFTSERIAL
feature GPS
feature TELEMETRY
feature RSSI_ADC
feature LED_STRIP
feature PWM_OUTPUT_ENABLE

# beeper
beeper -ON_USB

# map

# serial
serial 0 1 115200 57600 0 115200
serial 1 2 115200 57600 0 115200
serial 2 16 57600 57600 57600 115200
serial 31 1 9600 38400 0 115200

# led
led 0 15,15:S:C:2
led 1 14,15:S:CW:2
led 2 9,15:S:CW:2
led 3 8,15:S:CW:2
led 4 7,15:S:CW:2
led 5 6,15:S:CW:2
led 6 1,15:S:CW:2
led 7 0,15:S:C:2
led 8 11,0:N:C:1
led 9 10,0:N:CW:1
led 10 9,0:N:CW:1
led 11 8,0:N:CW:1
led 12 7,0:N:CW:1
led 13 6,0:N:CW:1
led 14 5,0:N:CW:1
led 15 4,0:N:C:1

# color

# mode_color

# aux
aux 0 2 1 900 1200
aux 1 4 1 900 1075
aux 2 4 1 1250 1350
aux 3 3 0 1100 1250
aux 4 3 0 1350 1475
aux 5 3 0 1600 1750
aux 6 3 0 1900 2100
aux 7 9 0 1250 1500
aux 8 8 0 1500 2100
aux 9 11 1 1500 2100
aux 10 12 2 1700 2100
aux 11 18 1 1500 2100

# adjrange

# rxrange

# master
set looptime = 1000
set gyro_sync = ON
set gyro_sync_denom = 1
set gyro_hardware_lpf = 188HZ
set gyro_lpf_hz = 100
set gyro_notch1_hz = 320
set gyro_notch1_cutoff = 200
set gyro_notch2_hz = 160
set gyro_notch2_cutoff = 100
set acc_hardware = MPU6050
set acczero_x = 48
set acczero_y = -31
set acczero_z = 12
set accgain_x = 4082
set accgain_y = 4078
set accgain_z = 4078
set align_mag = CW270FLIP
set mag_hardware = HMC5883
set magzero_x = 116
set magzero_y = -54
set magzero_z = -17
set baro_hardware = MS5611
set blackbox_rate_denom = 16
set motor_pwm_rate = 2000
set motor_pwm_protocol = ONESHOT125
set failsafe_procedure = RTH
set nav_extra_arming_safety = OFF
set nav_auto_speed = 600
set nav_manual_climb_rate = 500
set nav_rth_altitude = 2000
set nav_mc_hover_thr = 1300
set stats = ON
set stats_total_time = 2496
set stats_total_dist = 10216

# profile
profile 1

set mc_p_pitch = 50
set mc_i_pitch = 50
set mc_d_pitch = 22
set mc_i_roll = 40
set mc_d_roll = 22
set dterm_lpf_hz = 50
set dterm_notch_hz = 160
set dterm_notch_cutoff = 80
set roll_rate = 60
set pitch_rate = 60
set yaw_rate = 60

# 
```

stronnag, 2017-07-01:
This uses @digitalentity's timer cleanup changes, so breakage is expected.  Kudos for finding the first breakage.

stronnag, 2017-07-01:
Also confirmed. My SPRF3 (firmware, on a DODO) works fine with serial RX, but there is no RX input when I switch to a PPM module.

digitalentity, 2017-07-02:
@krzysztofmatula @stronnag thanks for finding the bug - will look asap.

digitalentity, 2017-07-02:
Can't confirm. PPM works fine on SPRacingF3 board and latest development (315925eb59a38144ef915fe33d6d5eb625eef094)

stronnag, 2017-07-02:
Try enabling non-PWM ESC modes (e.g. OS125). Then there must be a timer clash, works OK after `defaults`, fails as soon as I enable OS125.  I suspect this is why it also worked for @giacomo892 on NAZE. 

Retested as above on 315925eb

giacomo892, 2017-07-02:
@stronnag Since on that quad I don't have an ESC that support OS125 I cannot personally try it. Sorry

The other one wich I use "in production" runs an old 1.7.1 with BARO and MSP FLOOD fix firmware and I am unable to test on it atm.

stronnag, 2017-07-02:
You don't need the hardware, just setting
````
set motor_pwm_protocol = ONESHOT125
````
stops PPM working. It was your success on NAZE that was the clue that there appears to be a timer clash (on at least NAZE and SPRF3).

digitalentity, 2017-07-03:
Bug found and eliminated, thanks everybody!

closed 2017-07-03 by digitalentity