iNavFlight/inav #1853: OSD Units

MartinHugh, 2017-07-02:
The function osdFormatDistanceStr() in nav/src/main/io/osd.c, appears to be used for both altitude and distance and uses a single setting (osdConfig()->units) to determine the units for both.

What would be really useful would be to be able to set units for each altitude / distance metric individually (this extends to speed also, but that is another issue).

In the UK (and in other countries) it is convention to measure height in feet and distance in metres (or nautical miles).   

What I would like to be able to do is set altitude in feet and distance in metres, as is possible in other flight controller OSDs.

Thanks 
(existing osdFormatDistanceStr() follows).

```
/**
 * Converts altitude/distance into a string based on the current unit system.
 * @param alt Raw altitude/distance (i.e. as taken from baro.BaroAlt in centimeters)
 */
static void osdFormatDistanceStr(char* buff, int32_t dist)
{
	int32_t dist_abs = abs(osdConvertDistanceToUnit(dist));

    switch (osdConfig()->units) {
        case OSD_UNIT_IMPERIAL:
	        if (dist < 0)
	            sprintf(buff, "-%d%c", dist_abs / 100, SYM_FT);
	        else
	            sprintf(buff, "%d%c", dist_abs / 100, SYM_FT);
	        break;
        default: // Metric
            if (dist < 0)
                sprintf(buff, "-%d.%01d%c", dist_abs / 100, (dist_abs % 100) / 10, SYM_M);
            else
                sprintf(buff, "%d.%01d%c", dist_abs / 100, (dist_abs % 100) / 10, SYM_M);
    }
}
```

moggiex, 2017-07-04:
**Totally agree**, duplicate of #1756 can someone add this to the "UK Mode" thread please, eg @digitalentity 👍 Matt

digitalentity, 2017-09-09:
Closed. Fixed in 1.7.3.

closed 2017-09-09 by digitalentity