iNavFlight/inav #1854: Pulses for Servo Midpoint

jstremmler, 2017-07-02:
I tried to replace the  NAZE32 FC on my Titus Tricopter by a (little) more modern  SPRACINGF3  FC but had to reinstall my NAZE.
What happened is that the SPRACINGF3 showed a very strange behavior for the servo (connected to PIN 7, motors 1-3 connected to Pin 1-3 of the FC) Although the servo_center_pulse rate was set to 1500 the FC showed an output of 1254 pulses to be seen for servo6 in the motor menue of the configurator. 

In contrast the NAZE showed  correctly 1500 pulses for servo 6 when servo_center_pulse was set to 1500.

Ok, I could have manipulated the pulses for the SPRACINGF3 by setting the servo_center_pulse rate to 1747, but servo movements were a little "strange" with the SPRACINGF3, so I decided to take no risk and reinstalled my NAZE.....

stronnag, 2017-07-02:
Strange. I don't believe this is a generic problem. Behaves as expected here (on two different vehicles for as long as I can recall). This is SPRF3.
[image]

fwiw [cli diff](https://transfer.sh/j4hJD/mini-tri-inav.txt) (digital servo alert!).

jstremmler, 2017-07-03:
Well I did a quick and dirty test if this is a generic iNav problem or a specific problem of my FC.

I installed  Betaflight 3.1.7 on the same !!! SPRF3 and pulses for servo no. 6 in the motor menue were exactly at 1500 pulses. 
Next I flashed back iNav 1.7.1 and got 1253 puses for servo no. 6. 

So is this really a specific problem of my FC?

digitalentity, 2017-07-03:
Please provide settings dump.

jstremmler, 2017-07-03:
Ok, here is the CLI file.

[Titus_CLI_SPRF3.txt](https://github.com/iNavFlight/inav/files/1118607/Titus_CLI_SPRF3.txt)

digitalentity, 2017-07-09:
Can't reproduce the bug on 1.7.2 and your dump:
[image]

Either the bug is no longer relevant, or we accidentally fixed it with latest changes to servo mixer by @DzikuVx

DzikuVx, 2017-07-28:
@jstremmler Since we can not reproduct, I'm closing this one. If issue still exists with INAV 1.7.2 and latest version of Configurator, please reopen and provide a configuration dump for further analysis

closed 2017-07-28 by DzikuVx