iNavFlight/inav #1855: Motor Test Mode Grayed-Out In 3D Mode (1.7.1)

KenImhof, 2017-07-02:
3D mode is now working great with version 1.7.1, but it is difficult to calibrate ESCs as the Motor Test Mode is grayed-out and cannot be selected to control the motors.

digitalentity, 2017-07-02:
I believe this is done for safety reasons to prevent motors from spinning up.
Also 3D mode is not recommended since navigation core still has no idea about negative thrust.

KenImhof, 2017-07-02:
Thanks for the really quick reply.  With 1.7.1, 3D mode is now working great.  I have not yet tried with GPS POS/RTH, as I am waiting for a new GPS unit.  It would be great if the motor test could work like betaflight starting at 1500ms (motors off) and allow the slider to be moved up and down for forward and reverse thrust.  This would allow the motors to be tested and allow for easy ESC calibration.

DzikuVx, 2017-07-02:
@KenImhof please move this request to Configurator project . This will be closed. Thanks

closed 2017-07-02 by DzikuVx