iNavFlight/inav #1856: Omnibus F3 Pro

madoboy, 2017-07-02:
Hi. After flashing Omnibus hex on FC it wont recognise the board anymore. I can flash back Betaflight and connects without issue. Thanks

DzikuVx, 2017-08-09:
@madoboy without any other data we are not able to help

madoboy, 2017-08-09:
Hi. What kind of data you need? Happy to help

DzikuVx, 2017-08-09:
Well, anything. I flashed mine few days ago and everything worked like a charm for example. Do you have anything connected to the board for example?

madoboy, 2017-08-09:
No. Literally it arrived from Banggood. I plugged in tested connection with
Betaflight then wanted to flash with Inav. It goes to DFU mode and i can
flash it. It writes that finished the verifying. But when it restarts board
only flashes and won't recognise. I can give you afternoon what patern is
flashes. If is helping something. I tested on other comp and with different
cables and drivers.
After this i need to press boot button to go in DFU mode and i can flash
with Betaflight.

padcom, 2017-08-21:
I also have a problem with this board and iNav. While flashing it won't even go past the erase phase.

closed 2017-10-06 by DzikuVx

madoboy, 2017-10-07:
Hi Mr.Pawel. I saw your video on YouTube about pitot tube. Can you share
how to connect it. And also this gonna help with RTH and AH modes throttle
management? Are you maybe member of INav Autopilot group on Facebook it has
about 1000 member.

2017. okt. 6. 10:24 ezt írta ("Paweł Spychalski" <notifications@github.com

>):

Closed #1856 <https://github.com/iNavFlight/inav/issues/1856>.

—
<https://github.com/iNavFlight/inav/issues/1856#event-1281425368>, or mute
the thread

DzikuVx, 2017-10-07:
@madoboy https://quadmeup.com/how-to-connect-apm-airspeed-sensor-pitot-tube-to-inav/

Facebook group with 1000 members about INAV? Let me look for it :)

madoboy, 2017-10-07:
> @madoboy <https://github.com/madoboy> https://quadmeup.com/how-to-
> connect-apm-airspeed-sensor-pitot-tube-to-inav/

https://www.facebook.com/groups/1309125302467027/

madoboy, 2017-10-07:
Thanks for the link. I'm using INav on Mini Talon. Now with pitot tube it
gonna help with RTH throttle management?

2017. okt. 7. 9:47 ezt írta ("Stifler Fpv" <mado1326@gmail.com>):

DzikuVx, 2017-10-07:
Not yet, ATM it is use informative

madoboy, 2017-10-07:
Thanks for informations. Please join in our group you can have lot's of
inputs and experiences from pilots to help your development.